arendst/Tasmota #1595: Pzem004 and mqtt

gajotnt, 2018-01-13:
Since theres no module type for the Pzem module, i set it up using the basic module and defining the apropriet GPIOs. All works well. But in the POW the mqtt mesage is like this:
`"00:47:58 MQT: tele/sonoff/ENERGY`
and with the setup i explained it gives me this:
`00:49:08 MQT: tele/sonoff/SENSOR`
how do i make it use the ENERGY topic instead of SENSOR?

Thanks :/

LarsNorgaard, 2018-01-14:
Having somewhat the same problem with the POW after latest firmware.
I'm able to control the device just fine, but not able to read any of the data. I did change my state to SENSOR, as it says in the latest changelog
`state_topic: "tele/sonoffpow/SENSOR"`

gajotnt, 2018-01-14:
Domoticz of course works just fine, HomeAssistant is the one not reading the values.
Even if i put the right topic from the pzem004
```
  - platform: mqtt
    name: "Energy Casa"
    state_topic: "tele/sonoff/SENSOR"
    value_template: "{{ value_json.Today }}"
    unit_of_measurement: "kWh"
  - platform: mqtt
    name: "Power Casa"
    state_topic: "tele/sonoff/SENSOR"
    value_template: "{{ value_json.Power }}"
    unit_of_measurement: "W"
  - platform: mqtt
    name: "Voltage"
    state_topic: "tele/sonoff/SENSOR"
    value_template: "{{ value_json.Voltage }}"
    unit_of_measurement: "V"
```

`01:42:59 MQT: tele/powapart/ENERGY = {"Time":"2018-01-14T01:42:59","Total":129,"Yesterday":5,"Today":0,"Period":5,"Power":315,"Factor":0.97,"Voltage":236,"Current":1.367}`

`01:43:01 MQT: tele/sonoff/SENSOR = {"Time":"2018-01-14T01:43:01","ENERGY":{"Total":353.973,"Yesterday":11.159,"Today":1.323,"Period":5,"Power":630,"Factor":0.91,"Voltage":242,"Current":2.870}}`

arendst, 2018-01-14:
The json message has changed indeed to make it consistent with other sensors. As temperature sensors  use the same layout you might find a hass solution there. People managed to get temp and humidity in hass.

LarsNorgaard, 2018-01-14:
Ah now  I get it :)

It has to be like this now:
```
- platform: mqtt
    name: "Voltage"
    state_topic: "tele/sonoffpow/SENSOR"
    value_template: "{{ value_json['ENERGY'].Voltage}}"
    unit_of_measurement: "V"
```

closed 2018-05-28 by stale[bot]

hnaeem07, 2020-05-11:
Great work, really good contribution.