arendst/Tasmota #1596: Sleep and TelePeriod

tobox, 2018-01-14:
I have a NodeMCU with a MH-Z14A, which works perfectly with the Wemos D1 Mini / MHZ19 configuration. To save power, I set Sleep to 250, and noticed that that the telemetry rate went down. Is this the intended behaviour?
```

00:00:00 Project sonoff CO2Sensor (Topic co2sensor, Fallback CO2Sensor, GroupTopic sonoffs) Version 5.11.1b
00:00:01 WIF: Connecting to AP1 ToBoX in mode 11N as CO2Sensor...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on CO2Sensor.local with IP address 192.168.207.85
07:52:57 MQT: Attempting connection...
07:52:57 MQT: Connected
07:52:57 MQT: tele/co2sensor/LWT = Online (retained)
07:52:57 MQT: cmnd/co2sensor/POWER = 
07:52:57 MQT: tele/co2sensor/INFO1 = {"Module":"WeMos D1 mini","Version":"5.11.1b","FallbackTopic":"CO2Sensor","GroupTopic":"sonoffs"}
07:52:57 MQT: tele/co2sensor/INFO2 = {"WebServerMode":"Admin","Hostname":"CO2Sensor","IPAddress":"192.168.207.85"}
07:52:57 MQT: tele/co2sensor/INFO3 = {"RestartReason":"Software/System restart"}
07:53:13 MQT: tele/co2sensor/STATE = {"Time":"2018-01-14T07:53:13","Uptime":0,"Vcc":2.980,"Wifi":{"AP":1,"SSId":"ToBoX","RSSI":94,"APMac":"60:E3:27:76:C4:BA"}}
07:53:13 MQT: tele/co2sensor/SENSOR = {"Time":"2018-01-14T07:53:13","MHZ19B":{"CarbonDioxide":955,"Temperature":25.0},"TempUnit":"C"}
07:53:42 CMD: Sleep
07:53:42 MQT: stat/co2sensor/RESULT = {"Sleep":"100 (100)"}
07:53:44 MQT: tele/co2sensor/STATE = {"Time":"2018-01-14T07:53:44","Uptime":0,"Vcc":2.995,"Wifi":{"AP":1,"SSId":"ToBoX","RSSI":96,"APMac":"60:E3:27:76:C4:BA"}}
07:53:44 MQT: tele/co2sensor/SENSOR = {"Time":"2018-01-14T07:53:44","MHZ19B":{"CarbonDioxide":960,"Temperature":25.0},"TempUnit":"C"}
07:53:49 CMD: TelePeriod
07:53:49 MQT: stat/co2sensor/RESULT = {"TelePeriod":"15"}
07:54:14 MQT: tele/co2sensor/STATE = {"Time":"2018-01-14T07:54:14","Uptime":0,"Vcc":2.994,"Wifi":{"AP":1,"SSId":"ToBoX","RSSI":96,"APMac":"60:E3:27:76:C4:BA"}}
07:54:14 MQT: tele/co2sensor/SENSOR = {"Time":"2018-01-14T07:54:14","MHZ19B":{"CarbonDioxide":960,"Temperature":25.0},"TempUnit":"C"}
07:54:32 CMD: Sleep 200
07:54:32 MQT: stat/co2sensor/RESULT = {"Sleep":"200 (200)"}
07:54:38 CMD: TelePeriod 10
07:54:38 MQT: stat/co2sensor/RESULT = {"TelePeriod":"10"}
07:54:41 MQT: tele/co2sensor/STATE = {"Time":"2018-01-14T07:54:41","Uptime":0,"Vcc":2.997,"Wifi":{"AP":1,"SSId":"ToBoX","RSSI":94,"APMac":"60:E3:27:76:C4:BA"}}
07:54:41 MQT: tele/co2sensor/SENSOR = {"Time":"2018-01-14T07:54:41","MHZ19B":{"CarbonDioxide":961,"Temperature":25.0},"TempUnit":"C"}
07:55:22 MQT: tele/co2sensor/STATE = {"Time":"2018-01-14T07:55:22","Uptime":0,"Vcc":2.997,"Wifi":{"AP":1,"SSId":"ToBoX","RSSI":92,"APMac":"60:E3:27:76:C4:BA"}}
07:55:22 MQT: tele/co2sensor/SENSOR = {"Time":"2018-01-14T07:55:22","MHZ19B":{"CarbonDioxide":960,"Temperature":25.0},"TempUnit":"C"}
07:56:02 MQT: tele/co2sensor/STATE = {"Time":"2018-01-14T07:56:02","Uptime":0,"Vcc":2.997,"Wifi":{"AP":1,"SSId":"ToBoX","RSSI":96,"APMac":"60:E3:27:76:C4:BA"}}
07:56:02 MQT: tele/co2sensor/SENSOR = {"Time":"2018-01-14T07:56:02","MHZ19B":{"CarbonDioxide":961,"Temperature":25.0},"TempUnit":"C"}

```

davidelang, 2018-01-14:
setting sleep high messes with the teleperiod timing.

Sleep is there, but it really doesn't work that well

known issue, with a "don't do that" answer

On Sat, 13 Jan 2018, tobox wrote:

> Date: Sat, 13 Jan 2018 23:02:29 -0800
> From: tobox <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f3a42b0d72dc21da170274b94a2b3dc65a6725d0c92cf000000011672c28
>     592a169ce1130747d@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: [arendst/Sonoff-Tasmota] Sleep and TelePeriod (#1596)
> 
> I have a NodeMCU with a MH-Z14A, which works perfectly with the Wemos D1 Mini / MHZ19 configuration. To save power, I set Sleep to 250, and noticed that that the telemetry rate went down. Is this the intended behaviour?
> ```
>
> 00:00:00 Project sonoff CO2Sensor (Topic co2sensor, Fallback CO2Sensor, GroupTopic sonoffs) Version 5.11.1b
> 00:00:01 WIF: Connecting to AP1 ToBoX in mode 11N as CO2Sensor...
> 00:00:07 WIF: Connected
> 00:00:07 DNS: Initialized
> 00:00:07 HTP: Web server active on CO2Sensor.local with IP address 192.168.207.85
> 07:52:57 MQT: Attempting connection...
> 07:52:57 MQT: Connected
> 07:52:57 MQT: tele/co2sensor/LWT = Online (retained)
> 07:52:57 MQT: cmnd/co2sensor/POWER =
> 07:52:57 MQT: tele/co2sensor/INFO1 = {"Module":"WeMos D1 mini","Version":"5.11.1b","FallbackTopic":"CO2Sensor","GroupTopic":"sonoffs"}
> 07:52:57 MQT: tele/co2sensor/INFO2 = {"WebServerMode":"Admin","Hostname":"CO2Sensor","IPAddress":"192.168.207.85"}
> 07:52:57 MQT: tele/co2sensor/INFO3 = {"RestartReason":"Software/System restart"}
> 07:53:13 MQT: tele/co2sensor/STATE = {"Time":"2018-01-14T07:53:13","Uptime":0,"Vcc":2.980,"Wifi":{"AP":1,"SSId":"ToBoX","RSSI":94,"APMac":"60:E3:27:76:C4:BA"}}
> 07:53:13 MQT: tele/co2sensor/SENSOR = {"Time":"2018-01-14T07:53:13","MHZ19B":{"CarbonDioxide":955,"Temperature":25.0},"TempUnit":"C"}
> 07:53:42 CMD: Sleep
> 07:53:42 MQT: stat/co2sensor/RESULT = {"Sleep":"100 (100)"}
> 07:53:44 MQT: tele/co2sensor/STATE = {"Time":"2018-01-14T07:53:44","Uptime":0,"Vcc":2.995,"Wifi":{"AP":1,"SSId":"ToBoX","RSSI":96,"APMac":"60:E3:27:76:C4:BA"}}
> 07:53:44 MQT: tele/co2sensor/SENSOR = {"Time":"2018-01-14T07:53:44","MHZ19B":{"CarbonDioxide":960,"Temperature":25.0},"TempUnit":"C"}
> 07:53:49 CMD: TelePeriod
> 07:53:49 MQT: stat/co2sensor/RESULT = {"TelePeriod":"15"}
> 07:54:14 MQT: tele/co2sensor/STATE = {"Time":"2018-01-14T07:54:14","Uptime":0,"Vcc":2.994,"Wifi":{"AP":1,"SSId":"ToBoX","RSSI":96,"APMac":"60:E3:27:76:C4:BA"}}
> 07:54:14 MQT: tele/co2sensor/SENSOR = {"Time":"2018-01-14T07:54:14","MHZ19B":{"CarbonDioxide":960,"Temperature":25.0},"TempUnit":"C"}
> 07:54:32 CMD: Sleep 200
> 07:54:32 MQT: stat/co2sensor/RESULT = {"Sleep":"200 (200)"}
> 07:54:38 CMD: TelePeriod 10
> 07:54:38 MQT: stat/co2sensor/RESULT = {"TelePeriod":"10"}
> 07:54:41 MQT: tele/co2sensor/STATE = {"Time":"2018-01-14T07:54:41","Uptime":0,"Vcc":2.997,"Wifi":{"AP":1,"SSId":"ToBoX","RSSI":94,"APMac":"60:E3:27:76:C4:BA"}}
> 07:54:41 MQT: tele/co2sensor/SENSOR = {"Time":"2018-01-14T07:54:41","MHZ19B":{"CarbonDioxide":961,"Temperature":25.0},"TempUnit":"C"}
> 07:55:22 MQT: tele/co2sensor/STATE = {"Time":"2018-01-14T07:55:22","Uptime":0,"Vcc":2.997,"Wifi":{"AP":1,"SSId":"ToBoX","RSSI":92,"APMac":"60:E3:27:76:C4:BA"}}
> 07:55:22 MQT: tele/co2sensor/SENSOR = {"Time":"2018-01-14T07:55:22","MHZ19B":{"CarbonDioxide":960,"Temperature":25.0},"TempUnit":"C"}
> 07:56:02 MQT: tele/co2sensor/STATE = {"Time":"2018-01-14T07:56:02","Uptime":0,"Vcc":2.997,"Wifi":{"AP":1,"SSId":"ToBoX","RSSI":96,"APMac":"60:E3:27:76:C4:BA"}}
> 07:56:02 MQT: tele/co2sensor/SENSOR = {"Time":"2018-01-14T07:56:02","MHZ19B":{"CarbonDioxide":961,"Temperature":25.0},"TempUnit":"C"}
>
> ```
>
>

tobox, 2018-05-14:
I have just received the message, that this topic is stale now. I looked into the code to find out if it was feasible to process the teleperiod messages based on the RTC values (which seem to be OK even if TelePeriod is active), but it is not trivial. 

Maybe one possible workaround would be to calculate a correction factor for increasing the tele_period counter variable, based on the current sleep value. The results would probably not be perfect, but much better than the unexpected current  behaviour.

ascillato2, 2018-08-25:
Hi,

As the root cause of this and other issues is SLEEP, we can put all them together in #3581 so as to track the problem and try to search a solution.

closed 2018-08-25 by ascillato2

ascillato, 2018-08-26:
Solved