arendst/Tasmota #1597: Upload sonoff.ino.bin via OTA

pjanco, 2018-01-14:
Hello,
I successfully replaced the ewelink firmware with the tasmota firmware using the sonOTA python script and everything worked well. Then I decided to create my own binary from source code. So I downloaded the actual arduino tastoma project, configured the Arduino IDE, compiled the sketch (without any changes in the source code yet) and then uploaded sonoff.ino.bin file using the tasmota web interface. Fimrware was uploaded successfully, without error, but my sonoff did not work after the reboot. It is now restarted every 5 seconds. I can see the sonoff is still connected to my home wifi and sometimes I can ping on it (1ping sucessfull/ 15pings lost). But sonoff is not reacheble via web interface.

My compiled binary file had 518,656 bytes. The original pre-compiled binary file has only 495 808 bytes. My binary is bigger than half the size of the flash memory.

Do you guys think the problem is my binary was too big for OTA?
Why is my own binary bigger then pre-compiled binary?
I'm not sure what I did wrong. Any ideas?

I have ordered usb/ftdi module and waiting for it. I can't read uart debug output for this moment.

reloxx13, 2018-01-14:
https://github.com/arendst/Sonoff-Tasmota/wiki

all your problems and questions are written down in the wiki.

pjanco, 2018-01-16:
Hello,
so i finally got a ftdi converter and I fix my sonoff pow device by uploading a new firmware via the uart bootloader. Previously, I made a mistake in configuring Arduino IDE.

I skipped this step from the manual: "Optional: Preparing to Send OTA".
This is a confusing, evidently this step is not-optional. I have to do this even if I do not want to use OTA_upload function in arduino IDE.

This is the only difference between my previous failed attemption of firmware upgrade.

closed 2018-01-20 by pjanco