arendst/Tasmota #1598: 4CH PRO, selflock, POWER2 always on after restart

jcoudrais, 2018-01-14:
Hello everybody,

Quite a good product and also great job to manage these sonoff !

Weeks after getting a dual and a 4ch pro, didn't understand my problem, it seems it is a specific problem...

get in selflock mode
Power1 with delay
Power2 normal mode, but as restarting or after wifi lost, always ON, can't understand why...
 MQTT is ok, retained mode ok, with MQTT offline the same...
Power3 and Power4 were normal and also retained the state before restart or reconnect

The Power4 was available, decided to get it for my light...
 Also OK, but can't understand why...

Any idea ?

Log with POWER2 and POWER4 OFF
12:09:14 MQT: tele/sonoff_4CH_portail/STATE = {"Time":"2018-01-14T12:09:14","Uptime":0,"Vcc":3.155,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"","RSSI":40,"APMac":""}}
12:09:53 MQT: Attempting connection...
12:09:53 MQT: Connected
12:09:53 MQT: tele/sonoff_4CH_portail/LWT = Online (retained)
12:09:53 MQT: cmnd/sonoff_4CH_portail/POWER = 
12:09:53 MQT: stat/sonoff_4CH_portail/RESULT = {"POWER2":"ON"}
12:09:53 MQT: stat/sonoff_4CH_portail/POWER2 = ON (retained)
12:09:53 MQT: tele/sonoff_4CH_portail/INFO1 = {"Module":"Sonoff 4CH Pro","Version":"5.11.1","FallbackTopic":"DVES_8583FE","GroupTopic":"sonoffs"}
12:09:53 MQT: tele/sonoff_4CH_portail/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff_4CH_portail-1022","IPAddress":"192.168.0.29"}
12:09:53 MQT: tele/sonoff_4CH_portail/INFO3 = {"RestartReason":"Software/System restart"}
12:09:54 MQT: stat/sonoff_4CH_portail/RESULT = {"POWER1":"OFF"}
12:09:54 MQT: stat/sonoff_4CH_portail/POWER1 = OFF (retained)
12:09:54 MQT: stat/sonoff_4CH_portail/RESULT = {"POWER2":"ON"}
12:09:54 MQT: stat/sonoff_4CH_portail/POWER2 = ON (retained)
12:09:54 MQT: stat/sonoff_4CH_portail/RESULT = {"POWER3":"OFF"}
12:09:54 MQT: stat/sonoff_4CH_portail/POWER3 = OFF (retained)
12:09:54 MQT: stat/sonoff_4CH_portail/RESULT = {"POWER4":"OFF"}
12:09:54 MQT: stat/sonoff_4CH_portail/POWER4 = OFF (retained)
12:10:01 MQT: tele/sonoff_4CH_portail/STATE = {"Time":"2018-01-14T12:10:01","Uptime":0,"Vcc":3.152,"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"jackawifi","RSSI":42,"APMac":"00:0F:66:31:1C:7C"}}

With POWER4 On and POWER2 OFF and restart
12:10:32 MQT: tele/sonoff_4CH_portail/STATE = {"Time":"2018-01-14T12:10:32","Uptime":0,"Vcc":3.153,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"","RSSI":42,"APMac":""}}
12:11:14 MQT: Attempting connection...
12:11:14 MQT: Connected
12:11:14 MQT: tele/sonoff_4CH_portail/LWT = Online (retained)
12:11:14 MQT: cmnd/sonoff_4CH_portail/POWER = 
12:11:14 MQT: stat/sonoff_4CH_portail/RESULT = {"POWER2":"ON"}
12:11:14 MQT: stat/sonoff_4CH_portail/POWER2 = ON (retained)
12:11:14 MQT: tele/sonoff_4CH_portail/INFO1 = {"Module":"Sonoff 4CH Pro","Version":"5.11.1","FallbackTopic":"DVES_8583FE","GroupTopic":"sonoffs"}
12:11:14 MQT: tele/sonoff_4CH_portail/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff_4CH_portail-1022","IPAddress":"192.168.0.29"}
12:11:14 MQT: tele/sonoff_4CH_portail/INFO3 = {"RestartReason":"Software/System restart"}
12:11:15 MQT: stat/sonoff_4CH_portail/RESULT = {"POWER1":"OFF"}
12:11:15 MQT: stat/sonoff_4CH_portail/POWER1 = OFF (retained)
12:11:15 MQT: stat/sonoff_4CH_portail/RESULT = {"POWER2":"ON"}
12:11:15 MQT: stat/sonoff_4CH_portail/POWER2 = ON (retained)
12:11:15 MQT: stat/sonoff_4CH_portail/RESULT = {"POWER3":"OFF"}
12:11:15 MQT: stat/sonoff_4CH_portail/POWER3 = OFF (retained)
12:11:15 MQT: stat/sonoff_4CH_portail/RESULT = {"POWER4":"ON"}
12:11:15 MQT: stat/sonoff_4CH_portail/POWER4 = ON (retained)
12:11:22 MQT: tele/sonoff_4CH_portail/STATE = {"Time":"2018-01-14T12:11:22","Uptime":0,"Vcc":3.152,"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"","RSSI":40,"APMac":""}}

davidelang, 2018-01-14:
your log seems to be showing that you are retaining messages.

enable logging to 'more detailed' and it will show you what exactly is happening 
after the wifi connection is lost.

David Lang

jcoudrais, 2018-01-14:
Thank you for your answer:
power2 was off before restart, and here is the log:
 I don't know how it received POWER2 cmnd/sonoff_4CH_portail/POWER2, Data Size 2, Data ON
 any idea ?

21:23:38 MQT: Connected
21:23:38 MQT: tele/sonoff_4CH_portail/LWT = Online (retained)
21:23:38 MQT: cmnd/sonoff_4CH_portail/POWER = 
21:23:38 MQT: Subscribe to cmnd/sonoff_4CH_portail/#
21:23:38 MQT: Subscribe to cmnd/sonoffs/#
21:23:38 MQT: Subscribe to cmnd/DVES_8583FE/#
21:23:38 RSL: Received Topic cmnd/sonoff_4CH_portail/POWER2, Data Size 2, Data ON
21:23:38 RSL: Group 0, Index 2, Command POWER, Data ON
21:23:38 MQT: stat/sonoff_4CH_portail/RESULT = {"POWER2":"ON"}
21:23:38 MQT: stat/sonoff_4CH_portail/POWER2 = ON (retained)
21:23:38 MQT: tele/sonoff_4CH_portail/INFO1 = {"Module":"Sonoff 4CH Pro","Version":"5.11.1","FallbackTopic":"DVES_8583FE","GroupTopic":"sonoffs"}
21:23:38 MQT: tele/sonoff_4CH_portail/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff_4CH_portail-1022","IPAddress":"192.168.0.29"}
21:23:38 MQT: tele/sonoff_4CH_portail/INFO3 = {"RestartReason":"Software/System restart"}
21:23:38 CFG: Saved to flash at FB, Count 6500, Bytes 1648

davidelang, 2018-01-14:
it probably received it from the MQTT server because it has a retained message.

There is a page in the wiki on poweronstate and retained messages that tells you 
how to check for this and clear it.

closed 2018-05-28 by stale[bot]