arendst/Tasmota #1600: Rolling shutters, how?

povolomattia, 2018-01-14:
Hi,
i have a SonOdd Dual R2 with the latest firmware.
In the "configuration" tab, i don't see the "rolling shutters" option, so i use the "39 SonOff Dual R2".
The problem is because i want two buttons, maybe is a good idea don’t use a switch ON/OFF, but a button with “if you touch the button One: the rolling shutters go up , button two, go down” , but only if you hold down the buttons, is correct?
How can I make this feature?
Thank you very much and sorry for my english, i'm italian :)

calcipher49, 2018-01-15:
- Connect relay 1 to down input and relay 2 to up input
- Compute opening time of your shutter and set Pulsetime1 and Pulsetime2 (check commands Wiki for conversion from seconds)
- Put SetOption14 1 to prevent both relays from being activated at the same time.
Then push Button1 or set Relay1 ON will close shutter
push Button2 or set Relay2 ON will open shutter

closed 2018-01-15 by povolomattia

Sunrise17, 2018-04-16:
@calcipher49 , is it possible to open shutters in half (for ex 50%)? or only fully closed and opened?

calcipher49, 2018-04-16:
Yes it's possible, knowing previous state and full opening time (sames as specifed with Pulsetime in seconds).
You must use Backlog and Delay commands.
For example, if you shutter is closed and total opening time is 40 seconds, you shoud use 20 seconds ON time to open it 50%, so you can send:
``Backlog Power2 ON; Delay 200; Power2 OFF

Sunrise17, 2018-04-16:
That's so good, thanks for the information. Is there some plugin in Homebridge to use this property with Homekit?

calcipher49, 2018-04-16:
Don't know, sorry

Sunrise17, 2018-04-16:
Ok thanks again for the information

stefanbode, 2018-07-02:
I have now implemented shutter/blinds support in my mod. Please take a look in the branch of stefanbode and the wiki

stefanbode, 2018-07-03:
The trick is to maintain the real actual position at any time inside the firmware and react on changes. Only with this capability the 20%, 50% stuff will work. It is easy to just open and close completely. In my mod I update the real position any 0.05 seconds. Any change during operation will work with this accurracy and therefore is quite precise. With the original firmware this is not possible.

Sunrise17, 2018-07-03:
Dear @stefanbode , thanks for the information, today i will flash and check, then i will let you know about feedback.

Sunrise17, 2018-07-03:
@stefanbode , i have tried your firmware with homebridge plugin mentioned above;
first i have adjusted to position to 52; you can see homebridge log below; As you have seen that it didn't stop in 52 then it automatically adjusted to 44 and then 100. 
    
    [2018-7-3 10:00:50] [Window] Sonoff link for moving blinds:  http://sht01-8027/cm?cmnd=backlog%20pulsetime2%2046;power2%20on;delay%2046;pulsetime2%2080
    [2018-7-3 10:00:50] [Window] Moving up 4.6s
    [2018-7-3 10:00:50] [Window] stat/sht01/POWER2 message received = ON
    [2018-7-3 10:00:50] [Window] Setting target position to 52
    [2018-7-3 10:00:50] [Window] Blinds are moving. You need to wait. I will do nothing.
    [2018-7-3 10:00:54] [Window] stat/sht01/POWER2 message received = OFF
    [2018-7-3 10:00:54] [Window] lastPosition = 44 PositionState = 1 currentTargetPosition = 44
    [2018-7-3 10:00:54] [Window] stat/sht01/POWER2 message received = ON
    [2018-7-3 10:00:59] [Window] stat/sht01/POWER2 message received = OFF
    [2018-7-3 10:00:59] [Window] lastPosition = 100 PositionState = 1 currentTargetPosition = 100

I have stayed like this a couple of minutes, then device started to switch ON/OFF by himself out of my control. You can see below log;

    [2018-7-3 10:01:57] [Window] stat/sht01/POWER1 message received = ON
    [2018-7-3 10:02:05] [Window] stat/sht01/POWER1 message received = OFF
    [2018-7-3 10:02:05] [Window] lastPosition = 0 PositionState = 0 currentTargetPosition = 0
    [2018-7-3 10:04:34] [Window] stat/sht01/POWER2 message received = ON
    [2018-7-3 10:04:42] [Window] stat/sht01/POWER2 message received = OFF
    [2018-7-3 10:04:42] [Window] lastPosition = 100 PositionState = 1 currentTargetPosition = 100
    [2018-7-3 10:04:55] [Window] stat/sht01/POWER2 message received = ON
    [2018-7-3 10:05:03] [Window] stat/sht01/POWER2 message received = OFF
    [2018-7-3 10:05:03] [Window] lastPosition = 100 PositionState = 1 currentTargetPosition = 100
[2018-7-3 10:05:10] [Window] stat/sht01/POWER1 message received = ON
[2018-7-3 10:05:18] [Window] stat/sht01/POWER1 message received = OFF
[2018-7-3 10:05:18] [Window] lastPosition = 0 PositionState = 0 currentTargetPosition = 0

Sunrise17, 2018-07-03:
I have closed the homebridge, and restart the Sonoff Touch, now it is not automatically closing or opening and homebridge plugin is pushing him to switch on/off automatically. I will be glad if you can find some solution also for homebridge plugin. Finally we should control shutters from some interface.

stefanbode, 2018-07-03:
your homebridge definition if far to complicated. With this configuration my fork will behave like the original. Hopefully. pulsetime1 and 2 should be 0
the sommand to move is "shutterposition" and a value  from 0-100.
Depending on the preferences shutteropenduration and shuttercloseduration everything is automatically counted.
Additional commands (not very smart) are shutteropen, shutterclose, shutterstop. For more information please read the Wiki and enhance it, if you find it to complicated.

Sunrise17, 2018-07-03:
pulsetime1 and 2 should be 0 in console or in config.json? i set up 0 in console but still slider response is buggy.

stefanbode, 2018-07-03:
Can you just download the actual version because I found an overflow this morning? Just to make sure the behavior is from this one. Also, you can check with 10 seconds up and down. Then you will not be effected by the overflow.

stefanbode, 2018-07-03:
in the config jason you just send the position. No Backlog, no pulsetimes, no relay1 on offf. nothing

stefanbode, 2018-07-03:
This is definitly the wrong was to move it to 58%

[2018-7-3 10:00:50] [Window] Sonoff link for moving blinds:  http://sht01-8027/cm?cmnd=backlog%20pulsetime2%2046;power2%20on;delay%2046;pulsetime2%2080

The correct one should be:
http://sht01-8027/cm?cmnd=shutterposition%2056

Sunrise17, 2018-07-03:
problem caused by the [homebridge plugin](https://libraries.io/npm/homebridge-tasmota-mqtt-motor)?? I have downloaded latest version now i am uploading to board. I will check it

stefanbode, 2018-07-03:
Ah, ok now I see. The plugin is not needed anymore. This was a workaround to get the functioality more or less integrated into homeassistant. I would assume you can now take the normal shutter from homeassistant, like I did with openhab. Additional there is a reporting with TELEPERIOD about the actual position, if you want a feedback. Value is 0-100

13:56:26 MQT: hm/status/sonoff/SENSOR = {"Time":"2018-07-03T13:56:26","ANALOG":{"A0":7},"SHUTTER":0}

Sunrise17, 2018-07-03:
I have uploaded and tried but result is same as you can see in the updated [video](https://youtu.be/SqUWQeUmihE). I want to use it in my Homekit interface because all my other devices are there.

salcabo, 2018-11-23:
I tried to flash @stefanbonde patch in my sonoff t1 2ch but it doesn't appear in hardware list. Any chance to make it work?

Miroslave11, 2019-01-08:
Hello, what is the reason that shutters is missing in the last sonoff firmware? Is some alternative?
I have double T1 and I would like to use it for shutters.
Thank you,
Miroslav

miruim, 2019-05-30:
Hello @stefanbode . Greetings from Portugal
I would like to make a question about a anoying thing that happen to my rolling shutters.
At the begining i put it where i define it to be 50% and the with shutter50percent i put the value that in my case is 63 or 64. The problem is ... when i open it the shutter stops where i defined it, but it the shutter is completly open and i put it on 50%  does not stop in the same place. Is there any trick or something that could be done ??

Regards

stefanbode, 2019-05-31:
Yes, typically the speed is different going up and down. Therefore you have two values openduration and closeduration. Measure both and put them in. Now your shutter stop always at the same point. See wiki how to do so.

miruim, 2019-05-31:
Hello again and thank you for your reply 

If you mean shuttercloseduration and shutteropenduration this has been done since begining and in my case open and close duration is exactly. 

⁣Enviado por BlueMail ​

stefanbode, 2019-05-31:
Ok, please explain more what happen if goto 50 then 25 then 50 then 25 then 50. is the 50 stable or drift this away as the 25 drift away. If the shutter did not run long enough to get the 50 please increase shuttercloseduration . Same experiment 50 then 75 then 50 again 75 again and 50 again. If it works for 25/50 it must work also for 50/75. i recently added a motorstart delay of 0.5sec,but only if the motor is running and switches direction. Should not impact you. If there is a problem with the motorstart the first experiment will shift the shutter to the opendirection and the second one to the closedirection. Then we have an issue with you shutter currently not covered by the firmware. Looking forward to get your results. Hopefully just an issue of the speedparameters.

stefanbode, 2019-05-31:
@miruim please can you open an issue at the StefanBode fork. So we do not spam the master branch here. Thanks.