arendst/Tasmota #1603: MQTT status update

rajkedda, 2018-01-15:
Hi
  I have flashed the latest version of the Tasmota onto my Sonoff Basic and using it on HomeAssistant.  How do I send get status update of my sonoff basic and update it on HomeAssistant.  Here is my config for sonoff as light:
- platform: mqtt
  name: "Sonoff Living Room"
  state_topic: "stat/sonoffdev1/POWER"
  command_topic: "cmnd/sonoffdev1/POWER"
  qos: 1
  payload_on: "ON"
  payload_off: "OFF"
  retain: true

and in configuration.yaml file for the recorder:
  db_url: mysql://userid:password@localhost/homeassistant?charset=utf8

When I use the HomeAssistant UI, everything works & maintains it fine.  But loses the status after reboot of Raspberry Pi or HomeAssistant.  

Any ideas on how to fix it?

pascalsaul, 2018-01-15:
https://github.com/arendst/Sonoff-Tasmota/issues/1497 (similair)

I'm facing the same issues...

A possible fix can be to enable retain but personally I don't like that option either. Had some sync issues with that option also. I think it is a good idea to implement this feature.

gajotnt, 2018-01-15:
Working great this way in my setup:
https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

```
switch:
  - platform: mqtt
    name: "Sonoff power"
    state_topic: "stat/sonoff/POWER"
    command_topic: "cmnd/sonoff/POWER"
    availability_topic: "tele/sonoff/LWT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: true
```

pascalsaul, 2018-01-15:
Yes, that's with retain what I was saying above ;)

I personally really don't like this option when you've a split brain situation. It sounds as a 'cool' funtion but when you're using it for complex situation it can cause unwanted situations. 

If a bulb is switched in your bedroom through your domotica and the power is retained = bulb on
Now an WiFi/MQTT/HA crash occurs = bulb is still on
I'm going to bed and switching the bulb off through manual control = bulb off
Your HA still thinks the bulb is on (so a sync issue) = bulb off
The WiFi/MQTT/HA service is fully restored = bulb on!
I'm getting awake because the bulb turned on by retain ;) = bulb off

Just a simple situation... that's way I don't want to use retain ;)

bensuffolk, 2018-01-15:
I have an automation rule that request the status of the device when home assistant starts. The standard configuration for the state topic on the light picks up the reply and updates the state.

Here is an example that queries the status of 3 x T1 switches, just extend it as needed for other switches / lamps etc.

```
- action:
  - data:
      topic: tasmota/cmnd/t1/office/POWER
    service: mqtt.publish
  - data:
      topic: tasmota/cmnd/t1/conservatory/POWER1
    service: mqtt.publish
  - data:
      topic: tasmota/cmnd/t1/conservatory/POWER2
    service: mqtt.publish
  alias: Query Status on startup
  trigger:
  - event: start
    platform: homeassistant

```

rajkedda, 2018-01-15:
Thank you.  I will take your suggestions and revert back if I have challenges.

issacg, 2018-01-15:
you can also modify teleperiod to get a status update more often (== faster when HA restarts)

bensuffolk, 2018-01-15:
If you update the telemetry period your causing more traffic through the MQTT server, which may or may not be an issue depending on the frequency you set and or the amount of devices you have. You also have to wait for the period to come around, whereas the automation I posted fires a request as HASS starts, so its going to be the fastest method of making sure it up to date.

However if your devices don't happen to be on line at the time, then they will miss the request. So you might want to also listen for the telemetry as well for a full belt and braces solution.

zibous, 2018-01-16:
Hi,

In addition, I have the switch still “PowerRetain On” set.
see: https://community.home-assistant.io/t/ha-sonoff-switch-does-not-work-properly/36500/5

issacg, 2018-01-16:
@bensuffolk in a home WiFi, I'd hope there's enough bandwidth to not have problems with less than many hundreds of devices, and most home routers will limit you to 254 or even 100 (based on common DHCP settings) devices, so I've always assumed that MQTT traffic is always negligible for the "classic" use-case.  Have you experienced differently?  If so, could you share (so I don't run into the same problems myself)?

bensuffolk, 2018-01-16:
@issacg I can't say I have run into any problems, just I tend to stick to the principal that if there is no need to put extra traffic on a network then I don't, thats all. I guess also you would be putting some extra load onto HASS as its having to process more data, but again with a few sensors its not going to be an issue.

I went for the ask for the data approach when you want it, but I see you can set the telemetry period as low as 10 seconds, so worst case if your listening in for that would be 10 seconds with the wrong state showing I guess.

Although the telemetry only gives power, not colour and dimmer etc for lights so you may still end up with incorrect state data for them if your only listening to telemetry changes. 

I think both options have merits, but combining them together is probably the best approach.

issacg, 2018-01-16:
@bensuffolk agreed completely.

fradducho, 2018-05-10:
I had the same problem and I solved it in the console by typing "PowerRetain = 1" so that the device can send the stat / xxxxx / POWER ON or OFF broker depending on its status

Is this action correct?

closed 2018-07-09 by stale[bot]

hybrixxx, 2019-08-10:
> I have an automation rule that request the status of the device when home assistant starts. The standard configuration for the state topic on the light picks up the reply and updates the state. [...]

thanks