arendst/Tasmota #1605: Sonoff Pow not running

asiersan, 2018-01-15:
I flashed sonof pow with the tasmota firmware. It conects to wifi but no web page shown. If i enter in ap mode to configure wifi, config web is shown but when return to normal mode same problem no config page.

I tried with diferent tasmota versions. So i think there is a problem with new pow devices or a new review???

ascillato, 2018-01-15:
Have you tried to increase SERIAL_LOG_LEVEL so as to DEBUG? Because when the device connects to WIFI it can show its IP address that commonly is different from the AP mode.

droneando, 2018-01-15:
Same issue with two diferents POWs previously flashed with previous versions. Also the relay is cicling  every 30 seconds aprox. Flashing back to 5.10.0 version and all working ok. 
5 Sonoff Basic flashed with 5.11.1 working ok.

asiersan, 2018-01-15:
ascillato the ip is diferent i know. I will try 5.10.0 version and say

asiersan, 2018-01-15:
same thing with 5.10 not runing web interface. In basic devices no problem.

If i press button 4 times apmode starts and i havce web interface to configure the wifi. But i see in the router that sonoff pow is conected but  no web interface.

The led blinks every 20s so i think is in reboot loop.

ascillato, 2018-01-16:
Hi @asiersan, I upload the Sonoff-Tasmota 5.11.1b firmware to a ESP8266 and set it up as SONOFF POW.

It works for me but as my device is a NodeMCU it has more memory, so may be that the problem you are experiencing is lack of memory. Have you tried the #define BE_MINIMAL on your firmware?

[image]

asiersan, 2018-01-16:
Ok @ascillato  with minimal version, is running. 
[image]

So i must run in minimal mode???

asiersan, 2018-01-16:
Ok now i reprogram with the sonoff.bin in web interface, and all runs... thanks.

ascillato, 2018-01-16:
Great. So was just a memory problem. Great that is working now for you ;)