arendst/Tasmota #1608: Webinterface does not work correct after a while

rqsnst, 2018-01-15:
I flash a sonoff Basic with version 5.11.1 
after a while the web Interface does not work any more
and i get the following output on Firfox:
'arTimeout(lt);}if(x!=null){x.abort();}x=new XMLHttpRequest();x.onreadystatechange=function(){if(x.readyState==4&&x.status==200){var s=x.responseText.replace(/{s}/g,"").replace(/{m}/g,"").replace(/{e}/g,"").replace(/{t}/g,"%'>
Sonoff Basic Modul
Sonoff

Sonoff-Tasmota 5.11.1 von Theo Arends`

ascillato, 2018-01-16:
That happened to me when I was having trouble with the 3.3V power. Did you check if that was ok?

stefanbode, 2018-01-16:
This can also happen if the device is low on memory. I do see the same effect. Please report your HEAP size. Especially when working with MQTT and web server this may happen. There is no real workaround. A reboot helps because the webserver claims memory need some time. Therefore there is some time when there is still enough memory. The only workaround is to disable MQTT. @arendst There is no memory leak. It is just that after starting with the webserver the memory goes down over some time to stay constant after ~10 minutes.

rqsnst, 2018-01-16:
@ascillato: did you mean the 3.3V ,during the flash process ?

@stefanbode : MQTT ist disabled, where i can see the heap size ?
the information menu shows folowing information:

ESP Chip ID 363552
Flash Chip ID 1327328
Real flash memory 1024kB
Total flash memory 1024kB
Used Flash memory 485kB
Authorized Flash Memory 516kB
Free RAM 27kB

rebooting helps me to but that is no option for me

ascillato, 2018-01-16:
So,  seems to be a just a memory problem in your case.
I was referring to the 3.3V regulator of the board  but was not the case for you. Great that now is working for you :)

ascillato, 2018-01-16:
I build with arduino my owns and works just fine. I have to try platformIO.

ScottyDS, 2018-01-16:
I am having a similar issue.
I have 2 Sonoff Basics, 1 PoW, 1 Dual, and 1 NodeMCU selected as D1 mini.
All running Tasmota 5.11.1.b

On Windows, With both Chrome and IE, Immediately after a flash, I can access the web gui.  10 minutes or so later however, the buttons on the page become grey, with similar text as in post 1.
5 minutes further, The page can no longer be accessed, but the browser does not time out.  Eventually a screen is presented with a lot of code similar to that in post 1 but no buttons.
MQTT still works perfectly, however the response of an external pushbutton on one of the Basics becomes retarded by about 5 to 10 seconds. So I assumed it was memory issues.  After 2 days of struggling with this,  I happened to boot my PC into Ubuntu Linux, and  found I could access the Web gui with Chrome.  I immediately booted back into Windows and found the gui to be broken.

The weird thing is that as soon as I managed to access the web gui from the Linux machine, the Pushbutton lag, also disappeared. When I went back to Windows and accessed the again broken gui, the button started lagging again. 

Not being able to read  code very well, I can only surmise, that when Tasmota was struggling to present a page to what I now believe may be a Windows issue,  Memory or CPU on the ESP gets chewed up. Note, A reboot or erase+reboot of the ESP does not fix the issue when browsing from Windows.

I tested 5.10.0c and found it have the almost the same behaviour. I hauled out an old Windows tablet PC that has not been patched in about 6 months, and it seemed to work without problems (Does not have Creator Fall edition of Windows)

Hope this helps to track down the issue.

ascillato, 2018-01-16:
@ScottyDS, for flashing, are you using arduino ide, platformIO or the compiled .bin files?

ScottyDS, 2018-01-16:
Arduino IDE.  Still trying to get to grips with platformIO.

ascillato, 2018-01-16:
Have you tried the [precompiled bin files](https://github.com/arendst/Sonoff-Tasmota/releases)?

ScottyDS, 2018-01-16:
Nope, Not yet.  Only just learned from this discussion that they were available.  will try them tomorrow.

webtop, 2018-01-16:
Pre-compiled binaries are usually the minimal requirements. A build is ALWAYS better because it looks at your specific environment and compiles based on that. If you can build, always build

ScottyDS, 2018-01-17:
I tested with the pre-compiled binaries, and can confirm that the same thing happens as with the built code.  The web page does not load on a Windows machine, but works fine on a Linux machine or an Android phone, all using Google Chrome.

TKroenert, 2018-01-19:
TL;DR
As workaround use prebuild-binaries oder add the following to your platformio.ini:
 platform = espressif8266@1.5.0 ; esp8266/Arduino version 2.3.0

Long:
I'm using S20 with prebuild Tasmota binaries and sonoff-OTA flashing for some time now, but yesterday I got a magic home LED-Stripe controller without OTA so I thought it would be a good starting point to setup a buildenvironment.
After installing VSC and platform.io I made a git clone , PlatformIO:Upload  and on the face of it, everything looks good. But then the strange things started:
- website with broken html-code  (starts with "lementById('t').in..." instead of "<html ..")
- no website (wireshark shows "TCP ACKed unseen segment", "TCP previous segment not captured")

The following configurations did NOT worked for me:
-  5.11.1 (git checkout 7ad979163f0af8af98947c595096b555afe1d1dc) and v1.6.0
-  5.11.1c (git checkout 1538d30c96a542d6126a6827ac94d45411058ffe) and v1.6.0

And the following configurations worked well:
-  5.11.1 (git checkout 7ad979163f0af8af98947c595096b555afe1d1dc) and v1.5.0
- prebuild binary sonoff.bin (v1.?.0, @arendst are you using v1.6.0?)

The hardware is a magic home LED-Stripecontroller containing an espmodule labeled ESP8285MOD  (with 9(!) pins on the right side).
PlatformIO: Home 0.6.1·Core 3.5.1, Installed Platforms: Espressif8266 v1.5.0 and v1.6.0

tobox, 2018-01-19:
Your findings are very similar to mine posted here:

https://github.com/arendst/Sonoff-Tasmota/issues/1499#issuecomment-355286267

But I think I remember seeing issues with v1.5.0 as well, but was not able to reproduce them.

rqsnst, 2018-02-11:
hi, i think i close this topic, cause the problem is solved for me flashing the bin file promoted on this site

closed 2018-02-11 by rqsnst