arendst/Tasmota #1609: Sonoff Basic/Tasmota/Home Assistant for Garage Doors - Issue with Entity State from GPIO14

Sky1980chs, 2018-01-15:
I have setup two Sonoff Basics (running Tasmota) to monitor status and control garage doors.

I use two reed relays connected to GPIO14 to determine door position/closed.

The Sonoff's are configured to switchmode 2, pulsetime 10 and switchtopic sensor1 & sensor2

I am using Home Assistant for monitor & control.

All is working great...except:

HA is not reliably reporting my GPIO14/reed relay status.
Example:
cover.double or cover.single is unknown

When I issue a command to open/close a garage door [cmnd/garage1/power], Tasmota/Sonoff
will publish state messages for both the Sonoff onboard relay and the GPIO14 

stat/garage1/POWER ON  (sonoff relay)
cmnd/sensor1/POWER1 OFF (reed relay)

1.  Why does GPIO14 use a Command (CMND) to publish status?
2.  How can I poll Sonoff to publish the status of GPIO14 on demand?
Example:
When I issue stat/sensor1/power1 the response is for the main Sonoff relay?  I do not get a response for GPIO14.

Sonoff/Tasmota publishes regular MQTT status updates that define the state of the GPIO14.

tele/garage1/SENSOR 
which includes "Switch1":"ON"

1.  Can the interval of this status update be changed?
2.  Is there a way to associate this status update information with HA entity states to then update the sensor status [cmnd/sensor1/Power1]?

Any examples you can provide would be appreciated as I am a newbie.

aquadat0r, 2018-04-23:
@Sky1980chs Did you get any answers to the above? I am specifically interested in "Why does GPIO14 use a Command (CMND) to publish status?"

closed 2018-06-22 by stale[bot]

jcconnell, 2018-07-09:
Can either of you share more info? I'm in the middle of configuring my Son off basic right now. I've modified the hardware and am working on the software and incorporating it into home Assistant.