arendst/Tasmota #1610: RF Bridge 433 has no IDX for the 16 Channels

Beliar-666, 2018-01-15:
Hello,
First tasmota is a great software.
Now my Problem. My T1 Dual has IDX Adressfields for the switches so i can easiely configure it for Domoticz.

The RF Bridge only has IDX fields for Sensors not for the 16 channels. So i can only switch them from domoticz using the http way.

Also a question of me is, if it is possible to send a signal to domoticz if the T1 or the Bridge is used from outside of domoticz? If i use the Remote, Domoticz says the Switch is off, but its already on. Same with the t1. If i switch via Webpage or directly th switch Domoticz will not recognize it.

Is there any way to fix the Problems?

Also i can not update the RF to the newest Tasmota. Tried with using the Minimal it says not enough memory.

Joeyhza, 2018-01-18:
Are you trying an OTA update on the RD Bridge when you hit the memory issue?

Also, seems for now, Tasmota firmware for the bridge doesn't have domoticz integration for the 16 channels. Switching them using the HTTP way works though as i'm sure you found out. I tried this option and abandoned it and opted to flash with Espurna firmware instead. It had IDX integration for the 16 channels. Will run my bridge that way for now and flash to tasmota if/when IDX on the 16 channels is implemented.

Beliar-666, 2018-01-20:
I tried to Update over the Webpage from the Bridge. Upgrade by web server wont do anything and File Upload hits the memory Issue.

Now i tried to flash Espurna but my Bridge wont let me. Every Time i try to it says:
`warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
`

Joeyhza, 2018-01-21:
OTA needs pretty much double the available memory available. Best option hardware flash it

closed 2018-06-04 by stale[bot]