arendst/Tasmota #1611: Dual Wall Socket outlet config

AmpacheUser, 2018-01-15:
I purchased one of [these](https://www.ebay.com/itm/Smart-WiFi-Wall-Socket-Plug-Home-Appliances-Remote-Control-App-Outlet-Plug/172935227616) Wall outlets from eBay and have successfully flashed Tasmota firmware on it. But I am not able to figure out how to control each outlet independently. I have tried configuring it as a Sonoff Dual or a Touch 2, but it only turns both plugs on with the first switch. Has anyone figured out of to use this outlet with Tasmota successfully?

ascillato, 2018-01-16:
Hi.
That wall outlet is very nice, I was looking for something like that.
Do you have a picture of the circuit? It has 2 relays or 2 solid state switches?

AmpacheUser, 2018-01-19:
Well, found [this](https://www.amazon.com/ask/questions/Tx29HYKS127PXE1/). Looks like the outlets cannot be controlled separately which kind of defeats the purpose of having 2 outlets. 
Thats why Tasmota cannot control them separately either. Oh well. Hopefully I can still find a use for it.

closed 2018-01-19 by AmpacheUser