arendst/Tasmota #1612: Sonoff config not saved

croc1234, 2018-01-15:
Hi, installed tasmota version 5.11.1 from source with USB upload to a Sonoff T1 3ch. After that i try to change the config in the web interface, but changes are not persistent. After a reboot the settings are gone. 
I downloaded the source files, changed the chg holder number and the wifi settings in the config and then compiled and uploaded the firmware to the sonoff device. Of course i did not set the minimal setting before compiling. Can somebody guide me in the right direction?

webtop, 2018-01-16:
I take it you have read the wiki? Like, a lot of it.

croc1234, 2018-01-16:
Hi, yes i have read the wiki and created the firmware according to the instruction https://github.com/arendst/Sonoff-Tasmota/wiki/Beginner-Guide---Create-your-own-Firmware-Build. Then uploaded with Atom to my Sonoff T1 3ch. I have searched the issues in the wiki as well and found https://github.com/arendst/Sonoff-Tasmota/issues/369 before i posted my previous message. That issue was closed though without a solution. The workarround proposed did not work for me. 
I have not selected the minimimal settings, but assumed that maybe it would have uploaded a minimimal version first. So I tried to upload a precompiled bin (full version) from github trough the webinterface. Reconfigured my wifi to the default "indebuurt1" SSID, connected to the device and tried to configure the device again. Unfortunately config is lost again after a reboot. I have no clue what i am doing wrong.

croc1234, 2018-01-16:
And webtop, thanks for you reply. I searched for persist in the wiki but found only information in "issues". Unfortunately nothing that helped me further at this point

stefanbode, 2018-01-16:
can you send some logfile? SERIALLOG 4 should be ok. If you have no seriallog try WEBLOG 4 in the web console. I assume the value will be written to the CFG_HOLDER but for whatever reason during reboot, the device decided to load the default values. on my devices, everything is still running perfectly.Only log files can help. you should also try a "RESET 2" to ensure the CFG_HOLDER is really empty.

croc1234, 2018-01-16:
Thank you for your suggestions Stefan. The log did give me more info. It seems that the buttons are triggered multiple times although i did not touch the buttons. When buttons are touched multipe times after each other the device resets itself as can be seen in the log.

10:43:36 APP: DrukKnop 2 meervoudig 1
10:43:40 RSL: Ontvangen topic /Reset, Data lengte 1, Data 1
10:43:40 RSL: Groep 0, Index 1, Opdracht RESET, Data 1
10:43:40 RSL: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
10:43:41 CFG: Gebruik standaardwaarden

Then i found in https://github.com/arendst/Sonoff-Tasmota/issues/773 the setting Setoption13 1 which disables the multiple touch/reset function and allows only single touch. I entered the command Setoption13 1 in the webinterface and now the device holds its config permanently. Thank you for your great help. Sonoff-tasmota rocks!

closed 2018-01-16 by croc1234