arendst/Tasmota #1613: Free program memory eliminating option to select device from webserver

ascillato, 2018-01-16:
First of all, amazing job you guys are doing here. Congrats! @arendst

Here is an idea to free some space for new features:

What about to eliminate the option on webserver of selecting the device. Just leave the configuration for it on the webpage. Because as that is hardware related, you are not going to change which hardware is on the fly. You should only need just to adjust the behaviour of the hardware and its relations with the outside world (home assistant, MQTT etc.)

You should be selecting your hardware on the code before uploading, like in sonoff.h

Also, could be added (if the user wants) a option to select language on the fly. This should be used only if you have a device with enough memory.

davidelang, 2018-01-16:
if you are using the same image on many devices, you really want to switch the 
module type after it's loaded. It doesn't save much to disable changing the 
module type (the cost is in supporting defineing the different GPIO types)

arendst, 2018-01-18:
The goal if tasmota is having one image for all devices so you'll need a tool to select it's functionality. You don't neeed the webserver at all as all chenges can be made using mqtt. So to save space the removal of the  webserver would be top priority.

Removing anything has become less of an issue with OtaMagic so expect bigger images in the future.

closed 2018-01-18 by ascillato