arendst/Tasmota #1619: Domoticz options doubt

gajotnt, 2018-01-16:
Hello, sorry for my ignorance, but are the "**Key idx**" in the domoticz page used for?

https://github.com/arendst/arendst.github.io/raw/master/media/domoticz3.jpg

webtop, 2018-01-16:
It’s a link to a device in Domoticz. In Domoticz, you creat a device, get its key idx, enter that into the module, and you’ll get data in Domoticz without anything else needing to be done. When I was using Domoticz it worked perfectly.

webtop, 2018-01-16:
The wiki page gives a walkthrough

gajotnt, 2018-01-16:
I know that, but I usually put the IDX where it shows in the image above. The "Key IDX" is the thing I don't know what's for.

gajotnt, 2018-01-16:
[image]

arendst, 2018-01-16:
I have sensor in my letterbox down the road. It switches three sonoff like devices to signal the postman rang once.

The letterbox domoticz index is filled in in all three sonoff like domoticz key idx to switch their lights on.

gajotnt, 2018-01-16:
I know you can have several devices with the same IDX.
What i dont get is the why theres a "**Idx 1**" and a "**Key Idx 1**". I usualy set my domoticz idx in the **Idx1** field, never used the "**Key Idx 1**" and dont understand for what it is used for, that is my doubt.

arendst, 2018-01-17:
I've tried to explain the use of Key Idx in my last answer. For some reason I failed...

gajotnt, 2018-01-17:
:( yeah i didnt get the logic of the Key Idx field.
I get it that the idx of the mailbox sonoff is the same as the other 3 sonoffs in the house, so when the postman rings some lights turn on in the house, and if you turn off the lights the signal should be reset then, this would be the way i would make this automation, maybe your using a better way and im not understanding the logic of the key idx.
The Wiki of domoticz doesent have anything about keys (only API Keys), Tasmota Wiki, only mentions this in the Known Commands but doesnt explain what it is.

arendst, 2018-01-17:
Actually another sensor connected to the backdoor is resetting the signal but that is out of scope here.

The main issue is that using the button on any of the three sonoffs will reset the signals on all three sonoffs. This could also be used in a long aile or staircase with different switches controlling a single light.

Once you get it you might want to update the Domoticz wiki page ;-)

gajotnt, 2018-01-17:
When i have some spare time ill try out to mess with the key idx and understand better its use and then for sure i will update the wiki :)

closed 2018-05-31 by stale[bot]