arendst/Tasmota #1620: BME680 found at 0x77 , but not recognized and not reporting

anpj123, 2018-01-17:
Hi, I connected BME680 to Sonoff Basic with Tasmota 5.11.1 . After MQTT command I2Cscan I get response {"I2CScan":"Device(s) found at 0x77"} , but the sensor is still not reporting. I tested also BMP180 and that one works perfect , but BME680 don't . I also tried 5.11.0 firmware with the same results .

arendst, 2018-01-17:
You'll need to enable bme680 support in user_config.h first. Then recompile and upload.

anpj123, 2018-01-19:
Perfect ! It works excellent now. Thank you very much. Your work on this firmware is remarkable.

closed 2018-01-19 by arendst

renne, 2018-03-17:
Hi, I connected the BME680 with SDA to GPIO2 and SCL to GPIO14 and selected "18 generic" for the ESP8266-12F, SDA for GPIO2 and SCL for GPIO 14. CS is connected to 3.3V and SDO to GND. Tasmota is compiled with `#define USE_BME680` uncommented in user_config.h.

I2CScan lists `00:51:18 MQT: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x76"}` but I can't find anything else about the BME680.

How can I debug this?

renne, 2018-03-17:
After switching the BME680 to address 0x77 by connecting SDO to 3.3V instead of GND the BME680 is detected. All sensor values are zero. Is there anything else to configure?

renne, 2018-03-18:
The BME680 works fine with the Pimoroni library on a Raspberry Pi with address 0x76.

renne, 2018-03-18:
Good news, The BME680 works. :-)
It was just a broken ground lane on the breadboard. :-(