arendst/Tasmota #1621: Sonoff POW: tele ENERGY wrong JSON format

curzon01, 2018-01-17:
v5.11.1 MQTT prefix "tele" JSON result for ENERGY is wrong:

Example for v5.11.1 tele ENERGY Json:
{"Time":"2018-01-17T10:49:50","ENERGY":{"Total":1.944,"Yesterday":0.054,"Today":0.037,"Power":10,"Factor":0.43,"Voltage":229,"Current":0.096}

In v5.10.0 it was correct:
{"Time":"2018-01-17T11:27:12","Total":1.945,"Yesterday":0.054,"Today":0.038,"Period":0,"Power":2,"Factor":0.09,"Voltage":231,"Current":0.077}

it might be that the change in 5.11.0
"Change Sonoff Pow Energy MQTT data message and consolidate Status 8 into Status 10"
is incompleted  :-)

arendst, 2018-01-17:
Strange mine look OK:
```
{"Time":"2018-01-17T11:34:19","ENERGY":{"Total":14.201,"Yesterday":0.064,"Today":0.023,"Period":0.2,"Power":1.9,"Factor":0.16,"Voltage":203.9,"Current":0.059}}
{"StatusSNS":{"Time":"2018-01-17T11:38:28","ENERGY":{"Total":14.201,"Yesterday":0.064,"Today":0.023,"Power":1.9,"Factor":0.18,"Voltage":207.1,"Current":0.052}}}
```
What's the output of your ``status 10`` message?

curzon01, 2018-01-17:
Very strange - to write this issue I previously downgraded to v5.10.0  using web interface to my single Sonoff POW.

I now manually upgraded again to v5.11.1  using the webinterface to answer your question - and now, believe it or not, it's solved =:-]

I noticed this issue due to my FHEM expandJSON module, which always complained the MQTT payload:
Bad JSON: SonoffPOW ENERGY: {"Time":"2018-01-17T10:33:06","ENERGY":{"Total":1.942,"Yesterday":0.054,"Today":0.034,"Power":10,"Factor":0.40,"Voltage":230,"Current":0.106}

Before I updated all my Sonoff (>10) with GroupTopic and "upgrade" - so I will close this issue.

I will try to repeat this situation and if it occurs again we can reopen issue.

curzon01, 2018-01-17:
it happened again - and unfortuanely - it might occuring _randomly_:

`19:01:56 MQT: home/tele/tasmota/SonoffPOW/SENSOR = {"Time":"2018-01-17T19:01:56","ENERGY":{"Total":1.956,"Yesterday":0.054,"Today":0.049,"Period":0,"Power":9,"Factor":0.43,"Voltage":233,"Current":0.088}} (retained)
19:02:00 MQT: home/tele/tasmota/SonoffPOW/UPTIME = {"Time":"2018-01-17T19:02:00","Uptime":8}
19:03:58 MQT: home/tele/tasmota/SonoffPOW/STATE = {"Time":"2018-01-17T19:03:57","Uptime":8,"Vcc":3.175,"POWER":"ON","Wifi":{"AP":1,"SSId":"wlan.project-home.info","RSSI":88,"APMac":"C0:25:06:FF:A4:3A"}}
19:03:58 MQT: home/tele/tasmota/SonoffPOW/SENSOR = {"Time":"2018-01-17T19:03:58","ENERGY":{"Total":1.957,"Yesterday":0.054,"Today":0.049,"Period":0,"Power":10,"Factor":0.53,"Voltage":233,"Current":0.077}} (retained)
19:05:00 MQT: home/tele/tasmota/SonoffPOW/MARGINS = {"PowerLow":"OFF"}
19:05:00 MQT: home/tele/tasmota/SonoffPOW/ENERGY = {"Time":"2018-01-17T19:05:00","ENERGY":{"Total":1.957,"Yesterday":0.054,"Today":0.049,"Power":16,"Factor":0.53,"Voltage":233,"Current":0.130} (retained)
19:05:02 MQT: home/tele/tasmota/SonoffPOW/MARGINS = {"PowerLow":"ON"}
19:05:02 MQT: home/tele/tasmota/SonoffPOW/ENERGY = {"Time":"2018-01-17T19:05:02","ENERGY":{"Total":1.957,"Yesterday":0.054,"Today":0.049,"Power":9,"Factor":0.44,"Voltage":233,"Current":0.089} (retained)
19:05:59 MQT: home/tele/tasmota/SonoffPOW/STATE = {"Time":"2018-01-17T19:05:59","Uptime":8,"Vcc":3.186,"POWER":"ON","Wifi":{"AP":1,"SSId":"wlan.project-home.info","RSSI":84,"APMac":"C0:25:06:FF:A4:3A"}}
19:05:59 MQT: home/tele/tasmota/SonoffPOW/SENSOR = {"Time":"2018-01-17T19:05:59","ENERGY":{"Total":1.957,"Yesterday":0.054,"Today":0.050,"Period":0,"Power":9,"Factor":0.48,"Voltage":233,"Current":0.084}} (retained)
19:08:01 MQT: home/tele/tasmota/SonoffPOW/STATE = {"Time":"2018-01-17T19:08:01","Uptime":8,"Vcc":3.187,"POWER":"ON","Wifi":{"AP":1,"SSId":"wlan.project-home.info","RSSI":86,"APMac":"C0:25:06:FF:A4:3A"}}
19:08:01 MQT: home/tele/tasmota/SonoffPOW/SENSOR = {"Time":"2018-01-17T19:08:01","ENERGY":{"Total":1.957,"Yesterday":0.054,"Today":0.050,"Period":0,"Power":10,"Factor":0.58,"Voltage":233,"Current":0.070}} (retained)
19:08:13 CMD: status 10
19:08:13 MQT: project-home/stat/tasmota/EG_ku_MS_Abzug/STATUS10 = {"StatusSNS":{"Time":"2018-01-17T19:08:13","ENERGY":{"Total":1.957,"Yesterday":0.054,"Today":0.050,"Power":9,"Factor":0.55,"Voltage":233,"Current":0.072}}}
19:10:02 MQT: home/tele/tasmota/SonoffPOW/STATE = {"Time":"2018-01-17T19:10:02","Uptime":8,"Vcc":3.187,"POWER":"ON","Wifi":{"AP":1,"SSId":"wlan.project-home.info","RSSI":84,"APMac":"C0:25:06:FF:A4:3A"}}
19:10:02 MQT: home/tele/tasmota/SonoffPOW/SENSOR = {"Time":"2018-01-17T19:10:02","ENERGY":{"Total":1.957,"Yesterday":0.054,"Today":0.050,"Period":0,"Power":9,"Factor":0.48,"Voltage":233,"Current":0.081}} (retained)`

Have a look at 19:03:58 - it's ok
at 19:05:02 it's wrong and the last char is missing.

The code above is copied from web interface console but it's the same on MQTT payload and on syslog

curzon01, 2018-01-17:
I retrieved my logs and it's true: The last char } for SENSOR JSON is missing every time the MARGINS event happened before!

davidelang, 2018-01-17:
I wonder if you are running into maxmessagesize limits

are the ones that fail longer than the ones that succeed?

arendst, 2018-01-17:
It's a reproducable bug! I will investigate.

jjlawren, 2018-01-19:
I don't see it mentioned in the comments, but the threshold topic is still 'ENERGY'. The POW tele format changed to 'SENSOR' with 'ENERGY' in the payload in 5.11. Shouldn't they be consistent?

MimbaMonkeyHouse, 2018-01-24:
I've got the same issue. Out of 12 flashed POWs, only the three that have not been updated to 5.11.1 are showing energy consumption. I guess they're going back to 5.10 for now...

closed 2018-02-11 by curzon01

chimeranzl, 2018-03-03:
5.12.0 appears to shows incorrect ‘current‘ consumption, power (watts) is correct, current too high!

curzon01, 2018-03-03:
@chimeranzl - can not confirmed:

### 5.10.0
```
MQT: dev/stat/tasmota/pow/STATUS2 = {"StatusFWR":{"Version":"5.10.0","BuildDateTime":"2017-12-01T13:55:52","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
MQT: dev/stat/tasmota/pow/STATUS8 = {"StatusPWR":{"Total":5.631,"Yesterday":0.068,"Today":0.011,"Power":9,"Factor":0.51,"Voltage":229,"Current":0.079}}
```
### 5.11.1i
```
MQT: dev/stat/tasmota/pow/STATUS2 = {"StatusFWR":{"Version":"5.11.1i","BuildDateTime":"2018-02-07T07:57:21","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
MQT: dev/stat/tasmota/pow/STATUS10 = {"StatusSNS":{"Time":"2018-03-03T12:16:33","ENERGY":{"Total":5.631,"Yesterday":0.068,"Today":0.010,"Power":9,"Factor":0.60,"Voltage":229,"Current":0.069}}}
```

### 5.12.0
```
MQT: dev/stat/tasmota/pow/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
MQT: dev/stat/tasmota/pow/STATUS10 = {"StatusSNS":{"Time":"2018-03-03T12:20:24","ENERGY":{"Total":5.631,"Yesterday":0.068,"Today":0.011,"Power":9,"Factor":0.35,"Voltage":228,"Current":0.078}}}
```

All are the same - try calibrate all three voltage, current and power before measurements

chimeranzl, 2018-03-03:
I get Current reading from the JSON string ENERGY.Current from the topic SENSOR as someone else mentioned above, for me the firmware 5.12.0 does not publish to ENERGY topic anymore unless I manually request it by sending a message to STATUS. See if the current reads correctly from the SENSOR topic for you?

curzon01, 2018-03-04:
Yes, topic SENSOR is the telemetry topic (periodically send by TelePeriod) for Pow and _Current_ value for topic SENSOR is also valid:

#### Firmware info
```
CMD: Status 2
MQT: dev/stat/tasmota/pow/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
```
#### MQTT SENSOR topic Pow ON
`dev/stat/tasmota/pow/SENSOR {"Time":"2018-03-04T14:56:18","ENERGY":{"Total":5.707,"Yesterday":0.077,"Today":0.010,"Period":0,"Power":9,"Factor":0.42,"Voltage":233,"Current":0.077}}`

#### MQTT SENSOR topic Pow OFF
`dev/stat/tasmota/pow/SENSOR {"Time":"2018-03-04T15:00:28","ENERGY":{"Total":5.707,"Yesterday":0.077,"Today":0.011,"Period":0,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}`

Anyway again and again: Calibrate all three values of Pow before going on...

chimeranzl, 2018-03-04:
```
openhab/sonoff/poolpump/STATE {"Time":"2018-03-04T23:15:06","Uptime":"2T19:50:17","Vcc":3.476,"POWER":"ON","Wifi":{"AP":1,"SSId":"iotbam","RSSI":52,"APMac":"06:18:D6:C2:05:0B"}}
openhab/sonoff/poolpump/SENSOR {"Time":"2018-03-04T23:15:06","ENERGY":{"Total":27.679,"Yesterday":9.391,"Today":8.577,"Period":5,"Power":851,"Factor":0.42,"Voltage":234,"Current":8.678}}
```

The above is what Im getting. It is most certainly is NOT consuming 8.6 amps - it is not even rated that high, the device (pool pump) that is plugged into it is on and consuming 851 watts which is correct, at 234 volts and 8.678 amps would imply the pump is drawing 2,030 watts which it is not.

guzzit3, 2018-03-04:
you have a power factor of .42 so i think the current would be about right.
pete

guzzit3, 2018-03-04:
.42 x 2030 = 852.6 Watts

chimeranzl, 2018-03-04:
The sonoff powers a controller that has an analog timer on it, and power off to the pool pump. The analog/rotary timer is set to turn on the pump between 8am and 6pm (only use sonoff for power monitoring, not on/off ability) When the timer is off, the pump is obviously off, but the current reading does not change - it stays at 8.6 amps. How can that be?

guzzit3, 2018-03-05:
The pump is an inductive load the current is out of phase with the
voltage.If you look at yours it has a power factor of 0.42 . The pump is
only producing 860 watts but the current needed to this will be much great
see previous calculation. It looks to me as the power metering is spot on.
Google power factor it will explain better.
Regards Peter

guzzit3, 2018-03-05:
Not sure how it can be still showing 860 watts with the pump off.
Thanks Pete

curzon01, 2018-03-05:
for me I capitulate  - nothing about my suggestion to calibrate to rule out e.g. absolutly wrong calculation base...