arendst/Tasmota #1622: Correct settings for compile but still getting miscompare error uploading.

nospam2k, 2018-01-17:
I'm using MacOS High Sierra, Arduino 1.8.5, Generic ESP8266 v2.4.0, 1M (no SPFFS), DOUT... Everything compiles and uploads fine on serial. I then try to upload and get the miscompare error. I previously had used the older 2.3 and 1M (64K SPFFS). Do I need to do anything else?

[image]

arendst, 2018-01-17:
Compile options are ok.

You can only upload a program if it's size is smaller than the amount of free space shown on the information webpage or with command status 4.

If space is too small upload the minimal version first and then the original version.

nospam2k, 2018-01-17:
Ok, thanks for that. I see that file size is a problem. Because I'm going to run into this with OTA on each update (I only did a default compile), is there a way to do optional includes? Ex. minimal compile option, --with xyz, conditional includes, etc.

arendst, 2018-01-17:
See wiki - what's new how to automate.

nospam2k, 2018-01-17:
Unless I'm missing something. I need to learn Visual Studio Code with Platform IO. Is there a way to compile a mimimal with Arduino IDE? I really don't want to use a m$ product as I'm sure it will eventually have to be paid for and no doubt a lot!

arendst, 2018-01-17:
For OtaMagic you'll need a webserver to serve the ota's as mentionend in the what's new.

To do it manually you can use arduino ide; compile the minimal once as you can use it the always; then compile your latest code.

You can upload using the webpage.

NB. Visual studio code is a rip off from the full Visual Code and runs just as well on Linux. It's free and once you know it you will never want to use Arduino ide again. My two cents.

nospam2k, 2018-01-17:
Ok, I'm pleasantly surprised how easy it was to install and use vscode and platformio on the Mac. It is just an extension available through vscode. I just opened the root folder of Tasmoto (didn't need to copy anywhere) and changed the platformio.ini and hit build and off it went. For some reason the first build failed but I just hit build again and it was successful. Uploaded via web page and then uploaded the full bin file. Done! Soo... thanks for your patience and quick responses. I'm looking forward to upgrading all of Sonoffs!

closed 2018-01-17 by nospam2k