arendst/Tasmota #1623: Have I screwed up my Sonoff devices?

SteveDinn, 2018-01-17:
I flashed tasmota onto 3 sonoff devices, and everything was working fine. But then I decided to play around a bit and I think I may have messed things up. Here's what I did.

1. I read the wiki and discovered all the things I could do.
2. I saw that the 'sleep' command could reduce power consumption.
3. I typed 'sleep 10' at the console on the web interface.
4. Without investigating the ramifications on that device, I did this to two more devices.

Now, I can still see my devices connecting to my router, but I can no longer connect to the web interface, nor does a port scan show anything on port 80.

I thought I had broken things horribly, so I tried reflashing one of the devices, but it had no effect. I can still see the device connecting to the wifi, but I cannot get to the web interface.

At this point, is there anything I can do to salvage the device? Is there any way I can set 'sleep' back to 0?

Thanks.

SteveDinn, 2018-01-17:
Another interesting symptom is the ping responses from the device. I assure you that my wireless network is otherwise very functional.

Reply from 10.1.1.126: Destination host unreachable.
Reply from 10.1.1.206: bytes=32 time=2118ms TTL=255
Reply from 10.1.1.206: bytes=32 time=98ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Reply from 10.1.1.206: bytes=32 time=673ms TTL=255
Reply from 10.1.1.206: bytes=32 time=2ms TTL=255
Request timed out.
Request timed out.
Reply from 10.1.1.206: bytes=32 time=2697ms TTL=255
Reply from 10.1.1.206: bytes=32 time=3ms TTL=255
Request timed out.
Request timed out.
Reply from 10.1.1.126: Destination host unreachable.
Reply from 10.1.1.206: bytes=32 time=2028ms TTL=255
Reply from 10.1.1.206: bytes=32 time=5ms TTL=255
Request timed out.

ddamron, 2018-01-17:
Possibly reflash them with tasmota again?

SteveDinn, 2018-01-17:
@ddamron I've tried reflashing. It exhibits the same behaviour post-flash. I've read that tasmota will keep configuration values after upgrading firmware. Could something be left behind after flashing?

davidelang, 2018-01-17:
reset the config (see the troubleshooting page for why flashing didn't do this)

ChrSch84, 2018-01-17:
Have the Same Problem with a Sonoff Basic with 5.11.1b with this Version i have ping no ping and so on.
With the old "Stable" Build there is no Problem. Same Settings / Same Atom Build Same PC.

0n3-70uch, 2018-01-18:
Hi, I notice the same with 5.11.1b. I go back to 5.10.0 and send you the result. But its also possible that we run into this problem:

https://www.reddit.com/r/Android/comments/7qb3k3/having_wifi_issues_lately_router_problems_linked/

ChrSch84, 2018-01-18:
Should write that i have this Problem with 1 ONE Basic Sonoff and 1 One Magic Control LED. I try to reflash direct with uart but it take a while because my devices are behind the wall 💃

SteveDinn, 2018-01-18:
After some experimentation, I managed to catch the web interface of one of my devices while it was briefly online, and enabled debug logging. I noticed the IPs of my Google Home devices connecting to it repeatedly, under the assumption that it was a WeMo (I had this emulation mode turned on).

> 17:46:51 WIF: Connected
17:47:03 UPP: WeMo Type 2, Response sent to 10.1.1.106:53713
17:47:11 WIF: Checking connection...
17:47:11 WIF: Connected
17:47:31 WIF: Checking connection...
17:47:31 WIF: Connected
17:47:37 UPP: WeMo Type 2, Response sent to 10.1.1.109:54092
17:47:44 UPP: WeMo Type 2, Response sent to 10.1.1.105:43709
17:47:51 WIF: Checking connection...
17:47:51 WIF: Connected
17:48:03 UPP: WeMo Type 2, Response sent to 10.1.1.106:53713
17:48:11 WIF: Checking connection...
17:48:11 WIF: Connected
17:48:31 WIF: Checking connection...
17:48:31 WIF: Connected
17:48:37 UPP: WeMo Type 2, Response sent to 10.1.1.109:54092
17:48:44 UPP: WeMo Type 2, Response sent to 10.1.1.105:43709
17:48:51 WIF: Checking connection...
17:48:51 WIF: Connected
17:49:03 UPP: WeMo Type 2, Response sent to 10.1.1.106:53713

So, I turned off WeMo emulation. I now believe this to be some errant behaviour caused by the combination of Google Home(s) on the network and WeMo emulation. I had the notion that it was because of the Google Home/Chromecast network bugs being reported lately, but ever since disabling emulation in tasmota, the issue *seems* to have gone away. We'll see if it lasts...it's only been 10 minutes :)

SteveDinn, 2018-01-18:
Closed in error.

closed 2018-06-01 by stale[bot]

cydia2020, 2018-06-10:
Same here, Alexa works fine but Google Home and Assistant just reject to pair with my S20 Smart Socket.