arendst/Tasmota #1625: Sonoff  Pulses relay on start-up

manipat, 2018-01-18:
Hello, keep having this issue that my Sonoff Basic for some reason briefly pulses on when it either reboots (via unplugging or reboot through the web server). This is potentially an issue for me as I intend to use it with a relay to activate a garage door. I've already turned off PowerRetain as well as set PowerOnState to 0 (off),  have it set up as a switch through mqtt on Home Assistant which also has PowerRetain flag set to false.

Here's my log, anyone have any idea?

00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff4-4728.local with IP address x.x.x.x
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/sonoff4/LWT = Online (retained)
00:00:06 MQT: cmnd/sonoff4/POWER = 
00:00:06 MQT: stat/sonoff4/RESULT = {"POWER":"ON"}
00:00:06 MQT: stat/sonoff4/POWER = ON
00:00:06 MQT: tele/sonoff4/INFO1 = {"Module":"Sonoff Basic","Version":"5.10.0","FallbackTopic":"DVES_667278","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/sonoff4/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff4-4728","IPAddress":"x.x.x.x"}
00:00:06 MQT: tele/sonoff4/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/sonoff4/RESULT = {"POWER":"OFF"}
00:00:06 MQT: stat/sonoff4/POWER = OFF
19:35:42 MQT: stat/sonoff4/RESULT = {"POWER":"OFF"}
19:35:42 MQT: stat/sonoff4/POWER = OFF

davidelang, 2018-01-18:
turn on more detailed logging to see where the power on command is being 
originated.

manipat, 2018-01-18:
00:00:06 MQT: Subscribe to cmnd/sonoffs/#
00:00:06 MQT: Subscribe to cmnd/DVES_667278/#
00:00:06 RSL: Received Topic cmnd/sonoff4/POWER, Data Size 2, Data ON
00:00:06 RSL: Group 0, Index 1, Command POWER, Data ON
00:00:06 MQT: stat/sonoff4/RESULT = {"POWER":"ON"}
00:00:06 MQT: stat/sonoff4/POWER = ON

hmm not sure where it's coming from? My home assistant definitely has retain off and qos set to 0

closed 2018-01-18 by manipat

davidelang, 2018-01-18:
check if your mqtt server has any retained messages (see the poweronstate/retain 
wiki page)

dony71, 2018-01-19:
@manipat : can you tell me what topic name you change to fix this issue? sonoff4 if your old topic name and new topic name is ?

manipat, 2018-01-31:
@dony71 Sorry for the late response, didn't see this message. But it used to be sonoff4 and I changed it to lgdr (for my garage door) after that, it stopped retaining the ON command and would start as OFF