arendst/Tasmota #1626: Support for Sonoff S31 with energy monitoring

zephyrr, 2018-01-18:
It would be great to support the new Sonoff S31 smart plug!

There are a number of cheap wifi controlled outlets for $20 or under, but the S31 is compact (two can fit on one US duplex outlet), supposedly capable of 15A, and most importantly **does energy monitoring**!
And several sources are offering it for under $17, shipped.

See the last photo on this page: [(https://fccid.io/2AE2J-S31/Internal-Photos/Internal-Photos-3692513)]

I'm guessing that it continues to use the ESP8266 (or ESP8285).

It looks from the FCC registration photo as if there are pads (perhaps thru hole if not filled with solder) for VCC, GND, TX, RX and D_TX, D_RX (whatever the latter are).

arendst, 2018-01-18:
Will be implemented once I receive mine.

phreakmonkey, 2018-01-22:
It is ESP8266ex based.  Sonoff-Tasmota runs fine on it.  I haven't tried getting the "energy monitoring" working yet, so I don't know what sensor it uses.

Here's photos to show how to disassemble it; it's not readily apparent when you get it:
[http://www.phreakmonkey.com/2018/01/sonoff-s31-disassemble-and-flash.html](url)

phreakmonkey, 2018-01-22:
Looking at the board I see the ChipSea 7766 energy monitoring chip.  Seems that it sends energy use via uart:
http://www.chipsea.com/UploadFiles/2017/08/11144342F01B5662.pdf

Sky1980chs, 2018-01-30:
Theo do you have any estimate on when you will receive a S31 and start working on the power features?  I can also confirm that your current firmware flashes no problem on S31 although no power measurements.
Pics. for reference:  https://community.blynk.cc/t/hacking-the-new-itead-studio-sonoff-s31/21532

arendst, 2018-01-30:
The S31 is still in the mail. Once received I have to look at integrating the ChipSea 776 functionality...

Sky1980chs, 2018-01-30:
CSE7766 reference:  http://www.chipsea.com/UploadFiles/2017/08/11144342F01B5662.pdf
http://www.chipsea.com/html/product/dnjs/201705/1878.html

arendst, 2018-02-01:
Got it. Now find time to add it...

dkgrizzly, 2018-02-02:
Very nice!

zephyrr, 2018-02-03:
Cool!  I had already ordered my own, glad to test when they arrive.

Sky1980chs, 2018-02-05:
Thank you Theo!

Sky1980chs, 2018-02-06:
Likely it is my error, but I can not get a successful build:
\sonoff\xdrv_03_energy.ino:391:27: fatal error: TasmotaSerial.h: No such file or directory

I have verified that the files exist in a subfolder /lib/TasmotaSerial 1.1.0/src

What am I doing wrong?

phreakmonkey, 2018-02-06:
@Sky1980chs - Repeat the steps for "setting up the Arduino IDE environment" from the Wiki.   Recently a new library was introduced and you need to copy it from the source tree /lib folder into the libraries/ folder for your Arduino setup.  You should probably copy the others while you're at it in case they've been updated since you first set up Sonoff-Tasmota

Sky1980chs, 2018-02-06:
Unfortunately, I saw phreakmonkey comments too late, but had a successful build anyways.
Will capture what worked for me in hopes it may help others:
1.  Copy TasmotaSerial-1.1.0 folder from the Tasmota github source /lib to the Arduino IDE install location /portable/libraries.  Did same for PubSubClient although I think only think changed was MQTT_MAX_PACKET_SIZE to 1000
2.  For the IDE configuration you need to change Flash size from 1M (64K SPIFFS) to 4M (1M SPIFFS)

arendst, 2018-02-06:
Hate to say but if I had the possibility to change all replies where people want to change Flash size from 1M (no SPIFFS) to any non supported value I would be very glad...

It's in the wiki, it's in the readme, it's in Theo's tips, it's everywhere but hey without a whip I can't force people to stay on the right path...

Sky1980chs, 2018-02-06:
Humm..  I tried flashing using 1M (64K SPIFF) and was not having success.  The unit returned unresponsive.  I tried several times.  I thought since the S31 has 4M flash I needed to increase.  When I did the S31 worked.  Coincidence?  
Are you advising that we should flash the S31 @ 1M (no SPIFF)?  I can try that...no problem.

arendst, 2018-02-06:
As you noticed it worked (once). Chances are the next OTA won't work. So stick to 1M (no SPIFFS) and report any problems on that config only.

Sky1980chs, 2018-02-06:
Hopefully this thread will help someone else and save you their questions.  As always, I really appreciate your work you do for this community!!

zephyrr, 2018-02-06:
Sky1980chs - 

> "Hate to say but if I had the possibility to change all replies where people want to change Flash size from 1M (no SPIFFS) to any non supported value I would be very glad..."

 could be a hint to edit your post above to remove the wrong info about setting 4M/1M, since arendst is regretting that he cannot do it.  (Thanks for the other tip and for trying to help others, tho).

Sky1980chs, 2018-02-06:
Hint accepted and previous erroneous comments removed to avoid confusion.

j2mc, 2018-02-09:
First, thank you!  It works great!  Second I have a quick question on the S31, will a short teleperiod cause any problems?  I'm thinking in the range of 30 to 60 seconds.

arendst, 2018-02-09:
No problem at all. You can go down to 10 seconds.

asmugala, 2018-02-15:
No matter what I do, I keep getting the attached error.  I have Arduino IDE setup for flashing a Sonoff Basic and it works great on those.  I got the following error on two different Sonoff S31's
[image]

j2mc, 2018-02-22:
Is anyone else getting random incorrect power readings from their S31?  I have 3 and 2 of them are randomly showing power readings that are not happening, this one shows 22,689 W at one point:
[image]

My other one is showing 150-170 W randomly with nothing plugged into it(unplugged dryer to test this problem):
[image]

It is also counting it in total power calcs, here message from console showing both yesterday and today over 3kWh used even though the dryer is unplugged:
`21:11:57 MQT: tele/dryer/SENSOR = {"Time":"2018-02-21T21:11:57","ENERGY":{"Total":19.69,"Yesterday":3.72,"Today":3.34,"Period":0,"Power":0,"Factor":0.00,"Voltage":124,"Current":0.000}}`
Graph of total power over last 24 hours:
[image]

oglodyte, 2018-02-25:
@arendst  - Thank you Theo, this is great addition! 

Is there a reason no user configurable GPIOs available for S31? Serials are easy to solder to, but are they used for energy monitoring?  

Wanted to add temp/humidity sensor to this little nice device.

Anyone discovered other GPOIs available on-board w/out soldering directly to the chip?

arendst, 2018-02-25:
First, serial is used to communicate  with the energy chip.
Second, Gnd is one of the AC wires again so stay away from the S31 inner parts when AC is conected.

oglodyte, 2018-02-25:
Thank you Theo, good point re mains and gnd.  
I assume  that it is not the case with S20 is it?  Or [this hack](https://github.com/arendst/Sonoff-Tasmota/issues/670) might be dangerous too.

j2mc, 2018-02-26:
Well, the S31 that was really acting up just stopped working completely.  It powers up and connects to wifi when connected I'm connect with the serial port, but no wifi or LEDs when under it's own power.  Any ideas to try and get it working again?

gxgani, 2018-03-05:
@asmugala did you happen to fix it yet? Having the same "warning: espcomm_sync failed" while uploading.

gxgani, 2018-03-12:
@asmugala thanks for getting back. In my case it was the PL2303, was not able to sync. Then I tried it with CH340G USB TTL, worked like a charm.

PS: PL2303 did flash S20 but was not able to flash S31.

jekkos, 2018-04-13:
@arendst did you get an EU vesion of this adapter? Or does the US version work on 220V as well?

jekkos, 2018-04-13:
Ok think I found my answer

Voltage Input: 90~264V AC, 50/60HZ

So probably just need to use US - EU adapter then

arendst, 2018-04-13:
The US version indeed works just fine on 240V using holiday power plugs :-)

The new pow R2 uses the same energy monitoring chip as the S31 so I would go for that.

nardev, 2018-04-21:
I  still didn't get modules so i can't test it, is energy monitoring implemented? possible to get values through mqtt?

jekkos, 2018-04-24:
Ok pow R2 arrived, any suggestions with regards to safety here? Is it ok to build my own socket and wire this thing in?

PiOver2, 2018-05-07:
I am having trouble flashing at least one Sonoff S31 with the Tasmota 5.13.1 from Arduino IDE.  
It seemingly flashes correctly via USB, and the button works to force WIFI_MANAGER and I can connect to the Web interface, but cannot login, and there is garbage on the UI.  Here is the HTML from the 192.168.4.1 interface

`<!DOCTYPE html><html lang="en" class=""><head><meta charset='utf-8'><meta name="viewport" content="width=device-width,initial-scale=1,user-scalable=no"/><title>������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������� - Configure WiFi</title><script>var cn,x,lt;cn=180;x=null;function eb(s){return document.getElementById(s);}function u(){if(cn>=0){eb('t').innerHTML='Restart in '+cn+' seconds';cn--;setTimeout(u,1000);}}function c(l){eb('s1').value=l.innerText||l.textContent;eb('p1').focus();}function la(p){var a='';if(la.arguments.length==1){a=p;clearTimeout(lt);}if(x!=null){x.abort();}x=new XMLHttpRequest();x.onreadystatechange=function(){if(x.readyState==4&&x.status==200){var s=x.responseText.replace(/{t}/g,"<table style='width:100%'>").replace(/{s}/g,"<tr><th>").replace(/{m}/g,"</th><td>").replace(/{e}/g,"</td></tr>").replace(/{c}/g,"%'><div style='text-align:center;font-weight:");eb('l1').innerHTML=s;}};x.open('GET','ay'+a,true);x.send();lt=setTimeout(la,2345);}function lb(p){la('?d='+p);}function lc(p){la('?t='+p);}</script><style>div,fieldset,input,select{padding:5px;font-size:1em;}input{width:100%;box-sizing:border-box;-webkit-box-sizing:border-box;-moz-box-sizing:border-box;}select{width:100%;}textarea{resize:none;width:98%;height:318px;padding:5px;overflow:auto;}body{text-align:center;font-family:verdana;}td{padding:0px;}button{border:0;border-radius:0.3rem;background-color:#1fa3ec;color:#fff;line-height:2.4rem;font-size:1.2rem;width:100%;-webkit-transition-duration:0.4s;transition-duration:0.4s;cursor:pointer;}button:hover{background-color:#0e70a4;}.bred{background-color:#d43535;}.bred:hover{background-color:#931f1f;}.bgrn{background-color:#47c266;}.bgrn:hover{background-color:#5aaf6f;}a{text-decoration:none;}.p{float:left;text-align:left;}.q{float:right;text-align:right;}</style></head><body onload='u()'><div style='text-align:left;display:inline-block;min-width:340px;'><div style='text-align:center;'><h3>Sonoff TH Module</h3><h2>�������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������</h2></div><form method='post' action='/'><br/><b>User</b><br/><input name='USER1' placeholder='User'><br/><br/><b>Password</b><br/><input name='PASS1' type='password' placeholder='Password'><br/><br/><br/><button>Ok</button></form><br/><div id='t' name='t' style='text-align:center;'></div><br/><div style='text-align:right;font-size:11px;'><hr/><a href='https://bit.ly/tasmota' target='_blank' style='color:#aaa;'>Sonoff-Tasmota 5.13.1 by Theo Arends</a></div></div></body></html>`

arendst, 2018-05-07:
S31 needs 1M(no spiffs) fot your next ota upgrade. It also works great on 1M(no spiffs).

All you had to do was resetting the settings with command reset 1 to fill a default friendlyname.

BigBill2000, 2018-06-14:
I am looking for information on how to add the power consumption values to Home Assistant. I am new to this, but have my S31 working in HA and I can turn it on and off. I am not sure what yaml commands to use to read the power. Any help?

blakeblackshear, 2018-06-14:
Here is what I use with mqtt: https://github.com/blakeblackshear/blakeshome-config/blob/master/home-assistant/packages/sensors/sonoff_s31.yaml

BigBill2000, 2018-06-15:
Thanks! Got it working now.

jpabst, 2018-06-18:
I flashed the latest firmware 5.14.0 to my S31 and it shows as Sonoff Basic Module.  I've tried 5.12.0 and reflashing to 5.14.0  (I flashed the minimal between each one)

Is there some secret to getting S31 to show as an S31?

ascillato2, 2018-07-10:
Hi,

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks.

closed 2018-07-10 by ascillato2

rikray02, 2019-01-31:
Hi,

I'm not an expert to compile with arduino, so, i can use esptool to upload bin file to Sonoff.  So to night, i upload a sonoff.bin files (6.4.1).  After that, all works fins but i can't see power information (voltage, current, etc.).

Any idea why?
Richard

rikray02, 2019-01-31:
Forget my last comment.  I'm very stupid.  I forgot to change Module Type.  Now, all works fine.  

Thanks a lot for all sonoff développement.  

Richard

xbliss, 2023-06-29:
> @asmugala thanks for getting back. In my case it was the PL2303, was just not able to sync. Then I tried it with CH340G USB TTL, worked like a charm.
> 
> PS: PL2303 did flash S20 but was not able to flash S31.

I am reading this thread but more confused. Does this S31 smart switch with energy monitoring have support across dual voltages? 

> The US version indeed works just fine on 240V using holiday power plugs :-)
> 
> The new pow R2 uses the same energy monitoring chip as the S31 so I would go for that.