arendst/Tasmota #1628: Sonoff touch 2 chanel

vMeph, 2018-01-18:
Hi could someone point me into the right direction in how to set up sonoff touch2 chanel?
Other then solder male pins to conect my ftdi do i need any more modifications?
To flash do i need any modificationson conf file other then set ssdi, pass etc.?
On sonoff touch 2 chanel do i hold down gpio0 to ground before plug usb or gpio2?

Any help would be apreciated
Thanks in advanced

davidelang, 2018-01-18:
on any ESP8266 based device, progrmming mode is activated by holding GPIO0 down 
when you power it up.

vMeph, 2018-01-18:
getting this error while trying to upload
`esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM3
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 36 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 36 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 36 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 36 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 36 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 36 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 36 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 36 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 36 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
`

in my com i get this
`
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2408, room 16 
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : DOUT
  SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

rf cal sector: 251
rf[112]`
>

closed 2018-01-21 by vMeph