arendst/Tasmota #1630: How to fix Home Assistant (Hass) discovery python exceptions

arendst, 2018-01-18:
Release 5.11.1c provides Hass Mqtt discovery for both switch and light. It works using the parameters in file xdrv_06_home_assistant.ino and I'm impressed with the ease of adding Tasmota devices to Hass this way.

I have two issues:

1) When I look at Hass logging there are several exceptions caused by parameter errors in Color and/or Brightness depending on the type of light (ColdWarm, RGB, RGBCW). While discovery seems to work these errors should not happen. Anyone with knowledge about Hass (and Python) might provide some info here for me to fix.  

2) Hass allows for effect selection (Tasmota Scheme) but I'm unable to get it working. It has something to do with Python String Lists and how Hass discovery wants to receive the Effect list in the JSON message. Again, if you know how to use it please provide the info here or add a PR.

emontnemery, 2018-01-21:
Related to MQTT discovery: Maybe default MQTT topic should be something more unique than "sonoff"?

arendst, 2018-01-21:
To address a device you need a unique topic. Users normally come up with bedroom, kitchen etc. I just use a name like sonoff and a numer. The fallback topic is unique anyway.

As the friendlyname will be th Hass name they will probably change that one to bedroom light.

So no need to make things more unique.

emontnemery, 2018-01-21:
What I'm after is to flash a device, and have it automatically discovered without too many manual steps.
This doesn't work if all devices use topic "sonoff".
How does the "fallback topic" work?

I agree the "Friendly name" does not need to be unique for the reason you mention.

davidelang, 2018-01-21:
you either need to use the fallback topic, or set the topic to something unique 
for your system.

when you are configuring things one at a time, having a known topic that the new 
device starts out with is great.

when you are mass configuring them, you need to use the fallback topics (and 
know what they will be) to configure them.

personally, I bring them up one at a time after flashing them so that I know for 
sure that the one I just powered up is the one I am working with. I've had too 
many cases over the years of 'autodetection' not identifying the devices 
properly, and even a rare case of "oops, that's not the device I though was 
there" takes so much time to resolve that I'm much happier with the slightly 
more manual process.

emontnemery, 2018-01-22:
Fallback topic will not be used by tasmota for sending tele information or the hass-style discovery, right?
I understand your explanation why manual configuration is beneficial in your case. At the same time, if using Hass-style Mqtt-discovery, there really is no need for any manual config of topic if the default topic is unique. How about making it a compile time option to use a unique default topic vs standard "sonoff" topic?

emontnemery, 2018-01-22:
@arendst What is the reason for function `GetMqttClient`; I don't get why it can't simply be replaced with:
snprintf(mqttclient, sizeof(mqttclient), Settings.mqttclient, ESP.getChipId());

arendst, 2018-01-22:
If you only knew what users would enter as mqttclient. More than a year ago people managed to invent names screwing up this line of code leading to exception. Simply replacing a %d by a %s is enough to make it crash.

There is more code in here to keep it running from user errors than functionality...

engrbm87, 2018-03-07:
@arendst I have been working on getting the schemes to work with Home Assistant. I think changing this line -- "\"effect_list\":\"[0, 1, 2, 3, 4]\"" -- to -- "\"effect_list\": [\"0\", \"1\", \"2\", \"3\", \"4\"]" -- will make it work. But it would be better if we can apply schemes by their names rather than number. 

I have done some modifications to the xdrv_01_light.ino file to be able to recognize the effect by its name when received from HA, which allowed me to change the effect list to -- "\"effect_list\":[\"Solid\", \"Wakeup\", \"Cycle_RGB\", \"Cycle_RBG\", \"Random\", \"Incandescent\", \"Fire\"]" -- and make it work but the problem is that i can't put the entire scheme list because of the size limit of the message i guess.

arendst, 2018-03-16:
@engrbm87 Thnx for the correct python syntax.

Using Theme names makes the total MQTT message too long for Tasmota as it is currently set for max 893 characters. 

It takes too much valuable ram using a longer string for just HA as all other messages are way shorter.

Would have been nice if HA was accepting multiple shorter mqtt messages...