arendst/Tasmota #1632: No "Sonoff Basic" Response after Flashing from Tasmota to ESPEasy

wmjengel, 2018-01-18:
I wonder if anyone else had this problem and came up with a solution. I tried to flash ESPEasy (minimal, then full using the Tasmota Configuration Tool), but my Sonoff Basic is now dead (no LED, no switching, nothing!). I'm pretty sure it flashed based on the ability to (originally access it and change the configuration using the setup screen. But the unit no longer responds. I don’t have a FTDI usb/serial adapter, so I was wondering if there was another way (OTA or jumpers) to reset the Sonoff Basic to default conditions so I can reload Tasmota again? Right now, pressing/holding the (black) button does nothing. I haven’t put any 5.5v connections on the unit, only the 110v power input, so I don't think I bricked it. Any thoughts?

SodoffBaldrick, 2018-01-19:
By default, ESPEasy doesn't show any LEDs and the button doesnt switch the relay. That's all done by using the web config and telling the firmware how you want it to react. I'm new to all this so theres every chance that someone more experienced will correct me on this, but Sonoff Tasmota is designed for the Sonoff devices (with the ESP chips) and so defaults the settings to "Sonoff Basic". It can be used on a range of non-Sonoff devices but the important thing is that it defaults to Sonoff Basic with the settings you'd expect, LED, button presses affecting the relay etc.
ESPEasy however is a generic firmware supporting many devices, one of which is the Sonoff. It's default configuration is *blank* so nothing happens by default.

What I'd suggest for you is that if it has flashed then you should check your WiFi to see if theres an open WiFi point, named...I believe...ESP_EASY_0. If not exactly that then it'll be obvious. If it is there then log on and configure it as you wish.

If it's not there, then I'd have thought that the firmware wasn't on there and that your only choice is to get a USB to Serial adapter and flash it from serial. For what it's worth, I'd suggest a CH340G instead of an FTDI. I've got both and find the CH340G to be more reliable. They're very cheap and the only way to flash a device that isn't online and already working.

wmjengel, 2018-01-19:
After rethinking what I did, after successfully completing the OTA upgrade to Tasmota, I used the Tasmota configuration software to flash ESPEasy (minimal). When that was successful (I got a message in the ESPEasy screen to upgrade from minimal, so) I reflashed with the 1024 ESPEasy bin file. That seemed to load too as evidenced by the various setup screens, including the MAIN info screen. However, when I couldn't get any response from the Sonoff Basic (no green LED, no manual switching, etc.), I read about a way to get the Sonoff to respond to Smartthings from the Tasmota setup, so I tried to flash from ESPEasy back to Tasmota. That is when the Sonoff stopped working. I lost the ESPEasy IP access, and now I can't get the Sonoff to respond (i.e., no wakeup LED flashes, no response on WiFi, etc.). So, from what I read in the response above, I guess I will need to get a USB to Serial adapter to manually reset/reflash the unit unless someone has discovered another way to get around this.

closed 2018-06-02 by stale[bot]