arendst/Tasmota #1634: MQTT_MAX_PACKET_SIZE error when compiling

ascillato, 2018-01-18:
What about to modify _sonoff.ino_ with the following code so as there is no need to go and modify other libraries?

**sonoff.ino, Line 41:**
```
// Max message size calculated by PubSubClient is (MQTT_MAX_PACKET_SIZE < 5 + 2 + strlen(topic) + plength)
#if (MQTT_MAX_PACKET_SIZE -TOPSZ -7) < MIN_MESSZ  // If the max message size is too small, throw an error at compile time. See PubSubClient.cpp line 359
  #error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 1000"
#endif
```
change to:
```
// Max message size calculated by PubSubClient is (MQTT_MAX_PACKET_SIZE < 5 + 2 + strlen(topic) + plength)
#if (MQTT_MAX_PACKET_SIZE -TOPSZ -7) < MIN_MESSZ  // If the max message size is too small, throw a warning at compile time. See PubSubClient.cpp line 359
  #warning "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increasing it to 1000"
  #undef MQTT_MAX_PACKET_SIZE
  #define MQTT_MAX_PACKET_SIZE 1000
#endif
```

@arendst Testing Tasmota v5.11.1c. It is working very good!!!!

arendst, 2018-01-18:
Did you test it with Arduino Ide too? I did a year ago and didn't work. Pls try too.

ascillato, 2018-01-18:
Yes, in Arduino IDE works:

[image]

It works just fine.

arendst, 2018-01-18:
Pls provide the output of command status 6

ascillato, 2018-01-21:
@arendst Where I can find the output of command status 6 ?

davidelang, 2018-01-21:
you can send the command via mqtt, serial, or the web console

ascillato, 2018-01-21:
Thanks @davidelang 

The output of Status 6 on the Console is:

[image]

[image]

So, it means that the proposed modification to _sonoff.ino_ works

I tested on Arduino IDE 1.8.2

Sergey-SRG, 2018-01-21:
[image]
[image]

Sergey-SRG, 2018-01-21:
Works in Arduino IDE and PlatformIO.

ascillato, 2018-01-21:
Closing #1634 as there is the PR #1657.

@Sergey-SRG, your PR is good enough for solving the issue, but if you like we can add the warning for the user.

Thanks! :+1:

ascillato, 2018-03-08:
Hi @arendst ,

As this warning works on Arduino IDE and also on Platform.io, can be added to Tasmota?

Thanks

closed 2018-03-13 by ascillato