arendst/Tasmota #1636: SendIR Problem

cinehorse, 2018-01-18:
I have a problem with IRsend and Tasmota/Mqtt (5.11.1)  on a sonoff basic (Sorry for my bad english)

- IR Diode is installed on GPIO14
- Select 08 IRSend
- Mqtt works Topic ;_ tasmota/irsend

Testing Szenario in the console : 
`irsend {"PROTOCOL":"SONY","BITS":20,"DATA":"691015"}`

so i get the follwing message:
`22:14:18 CMD: irsend {"PROTOCOL":"SONY","BITS":20,"DATA":"691015"}
22:14:18 MQT: tasmota/irsend/stat/RESULT = {"IRSend":"No PROTOKOLL, BITS or DATEN"}`

the same result  when i send from broker directly thid teststring:
`mosquitto_pub -q 2 -t tasmota/irsend/cmnd/irsend  -m '{"protocol": "SAMSUNG","bits": 32, "data": 3772793023}'
`

Whereis the problem?  
can you help me?

arendst, 2018-01-19:
Took a while before I got it but the problem is with the language. It should have been in English but apparently the German translation is used. So for now you should be able to execute the Sony command as
```
irsend {"PROTOKOLL":"SONY","BITS":20,"DATEN:"691015"}
```
I will change it in the next (pre)release to English as it should be.

cinehorse, 2018-01-19:
Yes, thats the solution , a simple translation bug 

So i change the json string  in the command  and get 
`{"IRSend":"Done"}` 

Thanks

closed 2018-01-19 by arendst