arendst/Tasmota #1638: H801 individual white control

Anroxo, 2018-01-19:
I have a H801 with an RGB and two white strips connected to it. Is there a way to control the white ones individually? Turn them on/off, adjust brightness. I guess Color and PWM mode don't work together.

emontnemery, 2018-01-20:
It's possible only by directly setting the color.
I guess what you want is three dimmers:
- 1 x RGB dimmer, for the RGB strip
- 2 x 1 channel dimmer, for the two white strips
I also think it would be very useful.

Anroxo, 2018-01-20:
Exactly. That would be great.

kylegordon, 2018-01-20:
That would be fantastic, controlling the warm and cold white outputs separately

ssirkakriss, 2018-02-13:
To have individual dimmers in Home assistant for a H801 My settings are:
[image]

And My HaSS config bit:
```# h801 test
light:
 - platform: mqtt
    name: "Test-H801-WW"
    state_topic: "stat/test-h801/RESULT"
    state_value_template: '{%- if value_json["PWM"]["PWM5"]| int > 0 -%} 1023 {%- else -%} 0 {%- endif -%}'
    command_topic: "cmnd/test-h801/PWM5"
    on_command_type: 'brightness'
    brightness_state_topic: "stat/test-h801/RESULT"
    brightness_command_topic: "cmnd/test-h801/PWM5"
    brightness_value_template: '{{value_json["PWM"]["PWM5"]}}'
    brightness_scale: 1023
    qos: 0
    payload_on: '1023'
    payload_off: '0'
    retain: true
    optimistic: false
    
 - platform: mqtt
    name: "Test-H801-CW"
    state_topic: "stat/test-h801/RESULT"
    state_value_template: '{%- if value_json["PWM"]["PWM4"]| int > 0 -%} 1023 {%- else -%} 0 {%- endif -%}'
    command_topic: "cmnd/test-h801/PWM4"
    on_command_type: 'brightness'
    brightness_state_topic: "stat/test-h801/RESULT"
    brightness_command_topic: "cmnd/test-h801/PWM4"
    brightness_value_template: '{{value_json["PWM"]["PWM4"]}}'
    brightness_scale: 1023
    qos: 0
    payload_on: '1023'
    payload_off: '0'
    retain: true
    optimistic: false
    
 - platform: mqtt
    name: "Test-H801-Red"
    state_topic: "stat/test-h801/RESULT"
    state_value_template: '{%- if value_json["PWM"]["PWM1"]| int > 0 -%} 1023 {%- else -%} 0 {%- endif -%}'
    command_topic: "cmnd/test-h801/PWM1"
    on_command_type: 'brightness'
    brightness_state_topic: "stat/test-h801/RESULT"
    brightness_command_topic: "cmnd/test-h801/PWM1"
    brightness_value_template: '{{value_json["PWM"]["PWM1"]}}'
    brightness_scale: 1023
    qos: 0
    payload_on: '1023'
    payload_off: '0'
    retain: true
    optimistic: false
 
 - platform: mqtt
    name: "Test-H801-Green"
    state_topic: "stat/test-h801/RESULT"
    state_value_template: '{%- if value_json["PWM"]["PWM2"]| int > 0 -%} 1023 {%- else -%} 0 {%- endif -%}'
    command_topic: "cmnd/test-h801/PWM2"
    on_command_type: 'brightness'
    brightness_state_topic: "stat/test-h801/RESULT"
    brightness_command_topic: "cmnd/test-h801/PWM2"
    brightness_value_template: '{{value_json["PWM"]["PWM2"]}}'
    brightness_scale: 1023
    qos: 0
    payload_on: '1023'
    payload_off: '0'
    retain: true
    optimistic: false
 
 - platform: mqtt
    name: "Test-H801-Blue"
    state_topic: "stat/test-h801/RESULT"
    state_value_template: '{%- if value_json["PWM"]["PWM3"]| int > 0 -%} 1023 {%- else -%} 0 {%- endif -%}'
    command_topic: "cmnd/test-h801/PWM3"
    on_command_type: 'brightness'
    brightness_state_topic: "stat/test-h801/RESULT"
    brightness_command_topic: "cmnd/test-h801/PWM3"
    brightness_value_template: '{{value_json["PWM"]["PWM3"]}}'
    brightness_scale: 1023
    qos: 0
    payload_on: '1023'
    payload_off: '0'
    retain: true
    optimistic: false
#-----------
```

And the resulting in HaSS:

[image]
[image]

Hope it Helps.

Anroxo, 2018-02-13:
Thanks for the information!

EvgenPavlyuchek, 2019-04-11:
> 
> 
> To have individual dimmers in Home assistant for a H801 My settings are:

hi
new version 6.5 tashoma have 3 gpio in interface 
GPIO0 GPIO2 GPIO3

can you help how to do multiple control for 3 strips now?