arendst/Tasmota #1640: External switch and PIR on sonoff basic not working

brianbrown1962, 2018-01-19:
Hi Folks. First, and foremost, thank you all for your development efforts on Tasmota!

I will try to be brief with my issue...

I am setting up some sonoff basics. My target configuration is they will have an external momentary push button on GPIO14 and a PIR on GPIO1 (or GPIO3). The push button should function exactly as the onboard button does and directly toggle the relay. The PIR should just send MQTT on/off messages to my broker and I'll take actions on that within my HASS. This is probably a pretty common setup I would assume.

I have read everything I can find out there on switchmode and switchtopic, and everything I can find about PIR integration, however, that doesn't mean I'm still not doing something wrong ;)

I have the button attached to GPIO14. I set GPIO14 to "09 Switch1" in the module configuration GUI. I set switchmode to 3 in the console with the "switchmode1 3" command. The button works perfectly, exactly as the onboard does.

Now the damn pir...I'm using a common HC-SR501. I've connected the signal pin to the TX pin of the sonoff basic, which is GPIO1. In the module configuration gui I have configured GPIO1 as "10 Switch2".
In the console, I have set "switchmode2 1". I see no response in the console when the pir is triggered.

I have verified that the pir is putting the signal pin high to 3.3 volts when it's triggered, and returns to 0 after the timeout.

I have tried setting switchtopic to 0 and 1. No change in the pir. If I change switchtopic to 3, or some custom name, the pir still doesn't register, and I lose the direct functionality of the button on gpio14.

I found too that if I do ".../cmnd/POWER" at the console, I get the proper status for the relay. If I do ".../cmnd/POWER1", I also get the correct status for the relay, however if I do a ".../cmnd/POWER2", I get a {"Command":"Unknown"} response. With the way I set things up, shouldn't I be having a POWER2 for the status of the PIR as switch2? That was my understanding from my reading of other examples.

I'm using version 5.11.1.

Is my configuration steps correct for what I want to accomplish? If not, please, please clarify it for me. If so, then I can try to further diagnose hardware issues but the hardware seems ok, but open to suggestions on that too.

Thanks very much.
Brian

brianbrown1962, 2018-01-19:
Update: It appears the issue was with the voltage supplied to the HC-SR501. Was trying to power it with 3.3 volts from the sonoff as per a couple of hacks found on the web. Apparently that doesn't work well ;) Powered the PIR with 5volts and the configuration now works as expected. So, I've ordered a couple of U3V12F5 3.3v to 5v step-up IC's and see if that will drive the 5v on PIR ok. So, for those looking to make the same configuration of external momentary button and pir, the above description works. Closing this issue. Thanks.

closed 2018-01-19 by brianbrown1962

reloxx13, 2018-01-29:
you can get 5v from the sonoff basic. i use this myself and it works.

[image]
[image]
[image]
[image]
[image]
[image]

alvsky, 2018-04-17:
Hi
I connected my pir sensor to GPIO1 and used 5V from sonoff basic as described.
I configured it as Switch3 and I get On Off messages.. 
BUT.. sensor switching On and Off every 30 seconds...

what am I missing here?

badgerhome, 2018-04-28:
@brianbrown1962 

> The push button should function exactly as the onboard button does and directly toggle the relay. The PIR should just send MQTT on/off messages to my broker and I'll take actions on that within my HASS

I am trying to do exactly this but I can only seem to do one or the other, how did you get this to work? 

Many thanks :)

badgerhome, 2018-04-29:
Thanks @reloxx13 but won't this change switch1 to switchtopic2 also and therefore not control the relay? Or can I set a switch topic per switch?