arendst/Tasmota #1641: SONOFF Basic relay blinks when changing Access Point

perozonp, 2018-01-19:
I have 14 sonoff basic, all of them configured to connect to two Access Points. They are supposed to always stay in AP1 which is a Raspberry PI 3 configured to be an Acces Point, but if that one fails (like when rebooting the raspberry pi) they should switch to AP2 which is my Comcast wireless router. Everything works fine except when after having the raspberry PI up and running again I send the mqtt comand to all of my sonoff devices to switch back to AP1: "cmnd/sonoffs/AP 1". This comand efectivelly switches them back to the raspberry pi AP but when doing that they also blink the relay that happens to be ON (not happening if the relay is off). Currently is not a big issue since the load are just lamps but not a good thing to happen if the load is something more sensitive (like a TV). I have tested this switching 1 by 1 and it doesn't blinks, but when I send the command for the group (all 14 at the same time) then this happens. Just a change in the AP should not cause the relay to blink.

arendst, 2018-01-19:
As changing AP's involves a device reboot the relay will click. As designed.

perozonp, 2018-01-19:
wow!, ok, thanks!
This will be challenging. I'll have to design some tricky automation to only send the AP change command to those devices that are OFF, and for those that are ON then send the AP change command upon detection on status change to OFF.

Got some "homework" to do over the weekend... ;)

closed 2018-01-19 by perozonp