arendst/Tasmota #1642: MQTT HSB color command

DerEnderKeks, 2018-01-19:
Do understand that right that there is no support to set a HSB color via MQTT right now? I found some code related to HSB but no way to use that with MQTT commands.

If so, it would be really nice to have this possibility as I'm using OpenHAB which can't natively send RGB values, but only HSB values. 
(The format should be like it's for RGB right now; "h,s,b")

ThomDietrich, 2018-01-21:
One can always build an openHAB rule to translate back and forth between HSB and RGBW. See: https://github.com/eclipse/smarthome/blob/master/bundles/core/org.eclipse.smarthome.core/src/main/java/org/eclipse/smarthome/core/library/types/HSBType.java#L81

Still I agree, it would be amazing if Tasmota also supported HSB natively.

DerEnderKeks, 2018-01-22:
I thought about doing that but it wouldn't feel very native to do it like that and that's why I dislike that way.

ThomDietrich, 2018-01-22:
Not sure what you mean. To expect everything always served in just the right bits is also a bit too much to ask. If you are still interested, I build working rules yesterday evening:

* From RGBW to HSB+W: https://community.openhab.org/t/sonoff-b1-rgbcw-wifi-bulb-w-sonoff-tasmota/34202/15
* From HSB+W to RGBW: https://community.openhab.org/t/sonoff-b1-rgbcw-wifi-bulb-w-sonoff-tasmota/34202/18

DerEnderKeks, 2018-01-22:
I will probably use the rules for now but having a native integration of HSB in Tasmota would still be nice to have. :D

ThomDietrich, 2018-01-22:
Yes indeed, I don't argue with that 😉 

Actually I'd also like the implementation of an `RGBType` in openHAB. It's surprising that none exists...

Cypris2010, 2018-01-30:
I also would really appreceate this.

ThomDietrich, 2018-01-31:
The working openHAB solution can be found here: https://community.openhab.org/t/light-color-rgb-rgbw-rgbcw-to-and-from-hsb-dimmer-conversion/39132

closed 2018-04-02 by arendst