arendst/Tasmota #1643: Calibration parameter for the temperature, the humidity and the air pressure

micropet, 2018-01-19:
First of all thank you for the great software, Theo.
I use it mostly on NodeMCU or Wemos D1 Boards.

I have a 2 feature requests:

1.
The measured values from the BME280 are therefore inaccurate.
I would like a calibration parameter for the temperature, the humidity and the air pressure.

2.
The Automatic Baseline Correction for the MH-Z19 also does not work well.

I would like to do a zeropoint calibration by command.

I do that with my own programs and it works perfectly.
I have 5 Co2 meters here and all show the same results.

Greetings Peter

arendst, 2018-01-20:
How often is that zero point calibration needed? Could I decide to do it once a day (too often perhaps) or once a week. Or is it only needed after some minutes after power on. 

I don't want to have another command so making this automatic would make me very happy.

micropet, 2018-01-20:
Hi Theo,

I only ever did the zero point calibration once.
For this, the sensor must be in free air for 20 minutes, ie outside.

After a few years you should calibrate again. The MH-Z19 uses a normal lamp as the light source. This does not shine so bright after years.

abs outputs different values every day. It depends on how much CO in the room is when abs is executed.

Greetings Peter

Frogmore42, 2018-01-21:
Here is an article on calibrating CO2 sensors.  https://www.co2meter.com/blogs/news/7512282-co2-sensor-calibration-what-you-need-to-know
Sounds like once a year would be okay for many uses.

micropet, 2018-01-21:
Yes, that will be enough in any case.
I think only abs bad. It depends on the ventilation behavior.
I have wrong values over and over again.

This is not the case with manual calibration.

chriskmn, 2018-02-06:
I would also love to have calibration value for temp, humidity etc. as my humidity values of the DHT22 are way too high.

tiberiuana, 2018-02-07:
👍 for the calibration parameter. 
The Wemos SHT30 shields also seem to measure way to high.

micropet, 2018-02-09:
Thank you for MH-Z19 manual calibration.
Now we need calibration value for temp, humidity, press etc. 
Then it would be perfect.

DonStefano, 2018-05-03:
I would like to see calibration for all sensors.

Frogmore42, 2018-06-17:
See https://github.com/arendst/Sonoff-Tasmota/issues/2966
For a way to do it.

ascillato, 2018-06-27:
Hi,

If your issue is solved, please close it. Thanks :+1:

closed 2018-06-27 by micropet

ThinkPadNL, 2018-08-10:
I see it has been implemented in the code  (mhz_cmnd_zeropoint) but how do i trigger this from Tasmota?