arendst/Tasmota #1644: TSL2561 not working Version 5.11.1 (Sonoff TH16)

joopy, 2018-01-19:
Hi I don't get a TSL2561 to work. I'm using GPIO 4 en GPIO 14

@ console

21:20:53 CMD: I2Cscan
21:20:53 MQT: stat/sonoffTH16-2/RESULT = {"I2CScan":"Device(s) found at 0x39"}

Anyone got a clue?

joopy, 2018-01-19:
#define USE_TSL2561                            // Add I2C code for TSL2561 sensor using library Adafruit TSL2561 Arduino (+1k2 code)

user_config.h

totally forgot that one...

joopy, 2018-01-19:
TSL2561 Illuminance | 0 lx
-- | --

A little further, but it keeps on 0 lx?

joba-1, 2018-02-03:
I had the same learning curve (didn't know I need to #define USE_TSL2561 and when I did, I got 0 lx).

The sensor has a floating address bit, So it should be found at 0x39. And the following code snippet does so in a separate sketch on a nodemcu (and reports proper lx after that);

`    Wire.beginTransmission(0x39);
    if( Wire.endTransmission() == 0 )
      Serial.print("found");
`
Tasmota reports always as 0x29, no matter if I set address bit to low, float or high.

>  lowvolt2 ESP-I2C: TSL2561 found at 0x29

for me it looks like boolean TSL2561::begin(void) has a problem. It probably should check result of Wire.endTransmission().

joba-1, 2018-02-03:
Hm, I set the address bit to low to actually make the sensors address 0x29. But still no joy.

joopy, 2018-02-04:
Thnx Joba, for now I give up. I'm not so into coding, so I will wait for someone to fix it. Mayby Arendst can help out.

Merdeka, 2018-02-04:
Just an idea.

Could this be an issue with the i2c bus speed?
standard mode: 100 kbit/s vs full speed: 400 kbit/s

Merdeka, 2018-02-04:
i2c bis speed wasn't the issue.

There seems to be a problem with the for loop in xsns_16_tsl2561:

for (byte i = 0; i < sizeof(tsl2561_addresses); i++) {
    tsl2561_address = tsl2561_addresses[i];
    tsl = new TSL2561(tsl2561_address);
    if (tsl->begin()) {
      tsl->setGain(TSL2561_GAIN_16X);
      tsl->setTiming(TSL2561_INTEGRATIONTIME_101MS);
      tsl2561_type = 1;
      snprintf_P(log_data, sizeof(log_data), S_LOG_I2C_FOUND_AT, "TSL2561", tsl2561_address);
      AddLog(LOG_LEVEL_DEBUG);
      break;
    }
  }

It only runs once and tries to start the tsl with address 0x29.
When I change the line tsl = new TSL2561(tsl2561_address); to tsl = new TSL2561(TSL2561_ADDR_FLOAT); my tsl starts working as it is on address 0x39.

joopy, 2018-02-04:
So we need to change that line off code and recompile?

arendst, 2018-02-04:
The problem is in the Adafruit library which fails to report a not present device. The solution above is a workaround to use a fixed address of TSL2561_ADDR_FLOAT bypassing other possible addresses.

If @Merdeka gets it to function with the float address then why can't @joba-1 get it to work with the default low address? Perhaps @Merdeka could try with the low address too. If it works I just settle for a fixed low/float/high address. You may decide.

NB. I do not own a TSL2561 (there are other device providing luminosity) so I can't test.

joba-1, 2018-02-04:
Hi, 
Not sure why it didnt work with low address, but this lib change makes autodetect work for me:

`diff -u Sonoff-Tasmota-5.11.1/lib/TSL2561-Arduino-Library/TSL2561.cpp Sonoff-Tasmota-5.11.1.tsl2561/lib/TSL2561-Arduino-Library/TSL2561.cpp 
--- Sonoff-Tasmota-5.11.1/lib/TSL2561-Arduino-Library/TSL2561.cpp       2018-01-07 22:45:40.000000000 +0100
+++ Sonoff-Tasmota-5.11.1.tsl2561/lib/TSL2561-Arduino-Library/TSL2561.cpp       2018-02-04 18:06:30.559187730 +0100
@@ -63,7 +63,9 @@
 #else
   Wire.send(TSL2561_REGISTER_ID);
 #endif
-  Wire.endTransmission();
+  if( Wire.endTransmission() != 0 ) {
+    return false;
+  }
   Wire.requestFrom(_addr, 1);
 #if ARDUINO >= 100
   int x = Wire.read();
`

arendst, 2018-02-05:
OK. I'll release the change in library next time.

joba-1, 2018-02-05:
Did some more research digging in the datasheet of the TSL2561.

It seems, to get a valid value from the chip, the command bit (0x80) has to be set in addition to the register itself (0x0a). An example in the current driver can be found e.g. in the enable() method.

If I dont do that, I get no valid values, just a reflection of the register number. 
E.g. if I just send TSL2561_REGISTER_ID (0x0A), i always get back TSL2561_REGISTER_ID (0x0A) again.
But if I send TSL2561_REGISTER_ID | TSL2561_COMMAND_BIT, then I get 0x50, at least if I wait long enough. For some time after powering up I get just 0x00.
I don't know why I get 0x50 and not 0x10 like described in the datasheet, but in the sparkfun driver example, also 0x5x is mentioned as expected value (without giving a reason for that).
My fix above, check address but not the id register (removing the check if (x & 0x0A) ...) seems to be the safest way to make it work.

That said: why only check for sensors on startup? The bus allows plugging them in and out any time...

joba-1, 2018-02-05:
Didnt see your test request, will check...

joba-1, 2018-02-05:
I can confirm, it works. Great! Here is the startup log:

`2018-02-05T17:48:44.238288+01:00 lowvolt2 ESP-DOM: idx 14, nvalue 0
2018-02-05T17:48:50.043828+01:00 lowvolt2 ESP-SHT: Sensor did not ACK command
2018-02-05T17:48:50.084088+01:00 lowvolt2 ESP-I2C: TSL2561 found at 0x39
2018-02-05T17:48:51.049465+01:00 lowvolt2 ESP-MQT: tele/lowvolt2/STATE = {"Time":"2018-02-05T17:48:51","Uptime":0,"Vcc":3.357,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"job.fritz.ssid","RSSI":78,"APMac":"00:18:39:C6:B6:32"}}
2018-02-05T17:48:51.156817+01:00 lowvolt2 ESP-MQT: domoticz/in = {"idx":13,"nvalue":0,"svalue":"65535"}
2018-02-05T17:48:51.160378+01:00 lowvolt2 ESP-MQT: tele/lowvolt2/SENSOR = {"Time":"2018-02-05T17:48:51","TSL2561":{"Illuminance":65535}}
`
As you can see, the driver has a bug (it checks for negative unsigned variable :) ). This results in 65535lx although it is very dark.
Two questions: the driver you use is obsolete. Why are you using the obsolete driver? There is a new version on github. And it might be even more compact because it reuses the adafruit sensor lib now.
Unfortunately this driver still has the same bug, but it might be easier to get it fixed there.
Would you mind switching to the new driver?

joba-1, 2018-02-05:
Btw. here is the fix (including the actually working id check):

```
--- Sonoff-Tasmota-5.11.1/lib/TSL2561-Arduino-Library/TSL2561.cpp       2018-01-07 22:45:40.000000000 +0100
+++ Sonoff-Tasmota-5.11.1.tsl2561-works/lib/TSL2561-Arduino-Library/TSL2561.cpp 2018-02-05 17:01:46.853477179 +0100
@@ -54,16 +54,21 @@
 }
 
 boolean TSL2561::begin(void) {
+  // Initialise I2C
   Wire.begin();
 
- // Initialise I2C
+  // Check if someone listens at _addr
   Wire.beginTransmission(_addr);
 #if ARDUINO >= 100
-  Wire.write(TSL2561_REGISTER_ID);
+  Wire.write(TSL2561_COMMAND_BIT | TSL2561_REGISTER_ID);
 #else
-  Wire.send(TSL2561_REGISTER_ID);
+  Wire.send(TSL2561_COMMAND_BIT | TSL2561_REGISTER_ID);
 #endif
-  Wire.endTransmission();
+  if( Wire.endTransmission() != 0 ) {
+    return false;
+  }
+
+  // Check if we are actually talking to a TSL2561
   Wire.requestFrom(_addr, 1);
 #if ARDUINO >= 100
   int x = Wire.read();
@@ -71,7 +76,7 @@
   int x = Wire.receive();
 #endif
   //Serial.print("0x"); Serial.println(x, HEX);
-  if (x & 0x0A ) {
+  if ( (x & 0xF0) == 0x50 ) {
     //Serial.println("Found TSL2561");
   } else {
     return false;
@@ -194,17 +199,18 @@
     {b=TSL2561_LUX_B8T; m=TSL2561_LUX_M8T;}
 #endif
 
-  unsigned long temp;
-  temp = ((channel0 * b) - (channel1 * m));
+  // Calculate luminosity of full light and ir-only channel in lux according to datasheet
+  channel0 *= b;
+  channel1 *= m;
 
-  // do not allow negative lux value
-  if (temp < 0) temp = 0;
+  // Only consider non-ir luminosity, avoid overflow
+  uint32_t lux = (channel0 > channel1) ? channel0 - channel1 : 0;
 
   // round lsb (2^(LUX_SCALE-1))
-  temp += (1 << (TSL2561_LUX_LUXSCALE-1));
+  lux += (1 << (TSL2561_LUX_LUXSCALE-1));
 
   // strip off fractional portion
-  uint32_t lux = temp >> TSL2561_LUX_LUXSCALE;
+  lux >>= TSL2561_LUX_LUXSCALE;
 
   // Signal I2C had no errors
   return lux;

```

joopy, 2018-02-06:
Just flashed 5.11.1i on the sonoff TH16, I now get 

TSL2561 Illuminance | 10708 lx

Isn't thats a little to much?

joba-1, 2018-02-06:
You made me wonder. I checked the code and datasheet:

The code simply uses getLuminosity() which returns raw sensor data, not lux!

I'll try to fix this...

joba-1, 2018-02-06:
If you want to check:

In file Sonoff-Tasmota-5.11.1/sonoff/xsns_16_tsl2561.ino change the line (around #70)

>     uint16_t illuminance = tsl->getLuminosity(TSL2561_VISIBLE);

to

>    uint32_t light = tsl->getFullLuminosity();
>    light = tsl->calculateLux(light & 0xFFFF, light >> 16);
>    uint16_t illuminance = (light > 0xFFFF) ? 0xFFFF : (uint16_t)light;

It compiles and shows a value for me that seems to make sense. But I have no equipment to verify correctnes.

joopy, 2018-02-06:
Yes joba-1 output is much more what i expected:

20:30:01 MQT: tele/sonoffTH16-2/SENSOR = {"Time":"2018-02-06T20:30:01","TSL2561":{"Illuminance":13}}  (light out / only monitor light)
20:35:04 MQT: tele/sonoffTH16-2/SENSOR = {"Time":"2018-02-06T20:35:04","TSL2561":{"Illuminance":95}} (light on semi direct light)

So yeah I think this is working!

joba-1, 2018-02-07:
It is working better, but not good yet.

1)
I sometimes get 0xffffffff from getFullLuminosity() raw value. Due to overflow in calculations within calculateLux() this results in 95lx. It should probably return an easily identifiable error value like 0xffffffff.
@arendst: Is there a convention for a sensor value in tasmota that says "value not valid"?

2)
Then, when raw brightness of visible + ir light rises above ~0x8000, it beginns to stall while the ir part still rises, until both values are equal at 0x90cc, resulting in calculated 54lx. Tested with a bright smartphone led, not sunlight.

3)
There is a define in the driver that changes lux calculations a bit (I think ~10-20%). I don't know how to determine which of the 2 variants is right for my chip.

All those are not tasmota errors, but tsl2561 driver errors. 
I see there is some clipping and auto gain  done in the newer version of the adafruit lib. I plan to try this out. Then there is a sparkfun lib, dont know if it is any better.

@arendst: Getting a reading with auto gain and integration time could take a second (maybe even a bit more). Would that be an acceptable response time for tasmota or should a "driver" do continuous readings so it has a proper value ready when asked for? The chip only uses 0.24mA typical when powered on, so that doesnt really matter compared to what the esp needs with active wlan.

Finally: I don't insist in using that chip. I just liked, I can read ir and normal light because in the end this might enable me to differ light sources (sun, led, ...). Anyone has a suggestion for an alternative?

arendst, 2018-02-07:
Current value not valid policy is do not display or keep it for up to 10 teleperiod reads then drop display.

I prefer reading time as low as possible. If driver can kick device to start processing, then back off and connects after a second again to read value that's fine with me.

joba-1, 2018-02-08:
to round this up here: 
1) The mystery about the two sets of parameters for lux calculation and the chip id 0x50 vs 0x10 are solved:
According to the newer datasheet (not the one adafruit links to, but this one, page 18: http://html.alldatasheet.com/html-pdf/560746/AMSCO/TSL2561/2958/19/TSL2561.html) 0x1? is the CS package and 0x5? the other ones. So one can choose the correct parameter set dynamically (not with #defines).
2) the programming error that checks for less than zero of an unsigned variable is already in the manufacturers example code - very reassuring :)
3) My pull request with the fix for the lux calculation is already in the dev branch. Thanks Theo! I plan to improve that with the new infos

joopy, 2018-02-10:
Just flashed my Sonoff Th16 with firmware 5.12a with TSL2561 plugged in I got following error.

00:00:00 Project sonoffth16-2 Sonoff (Topic sonoffTH16-2, Fallback DVES_0DD48D, GroupTopic sonoffs) Version 5.12.0a
00:00:00 WIF: Connecting to AP2 amor in mode 11N as sonoffTH16-2-5261...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on sonoffTH16-2-5261.local with IP address 192.168.0.94
10:17:21 MQT: Attempting connection...
10:17:23 MQT: Connected
10:17:23 MQT: tele/sonoffTH16-2/LWT = Online (retained)
10:17:23 MQT: cmnd/sonoffTH16-2/POWER = 
10:17:23 MQT: tele/sonoffTH16-2/INFO1 = {"Module":"Sonoff TH","Version":"5.12.0a","FallbackTopic":"DVES_0DD48D","GroupTopic":"sonoffs"}
10:17:23 MQT: tele/sonoffTH16-2/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffTH16-2-5261","IPAddress":"192.168.0.94"}
10:17:23 MQT: tele/sonoffTH16-2/INFO3 = {"RestartReason":"Fatal exception:3 flag:2 (EXCEPTION) epc1:0x4022d46a epc2:0x00000000 epc3:0x00000000 excvaddr:0x40269f20 depc:0x00000000"}
10:17:23 MQT: homeassistant/switch/sonoffTH16-2/config =  (retained)
10:17:24 MQT: stat/sonoffTH16-2/RESULT = {"POWER":"ON"}
10:17:24 MQT: stat/sonoffTH16-2/POWER = ON
10:17:25 MQT: stat/sonoffTH16-2/RESULT = {"POWER":"ON"}
10:17:25 MQT: stat/sonoffTH16-2/POWER = ON

Joba his fix:
In file Sonoff-Tasmota-5.11.1/sonoff/xsns_16_tsl2561.ino change the line (around no 70)

Was working fine for me.

arendst, 2018-02-10:
Strange as his fix should be in the latest versions too. Can you verify if that's the case?

joopy, 2018-02-10:
Arend around line 70 in xsns_16_tsl2561.ino

There is no such thing as:

uint32_t light = tsl->getFullLuminosity();
light = tsl->calculateLux(light & 0xFFFF, light >> 16);
uint16_t illuminance = (light > 0xFFFF) ? 0xFFFF : (uint16_t)light;

arendst, 2018-02-10:
See https://github.com/arendst/Sonoff-Tasmota/pull/1825

joopy, 2018-02-11:
Thnx Arend, however it's not working for me. Joba is latest firmware working for you?

joba-1, 2018-02-11:
Hi joopy,

sorry, have not checked yet, only a patched 5.11.1. And that worked fine.
It could take a little time until I can test this.

Currently I work on my own Tsl2561 library. If we really want to use the
chip, we will need automatic gain, and maybe more accuracy than 1 lux.

Most is already done. Tipps welcome.
See https://github.com/joba-1/Joba_Tsl2561 if you like.

joopy, 2018-02-12:
Thnx Joba,

Looking forward. For now I will wait for a fix no hurry. When your library is finished it will be implemented in Tasmota?

joba-1, 2018-02-20:
Hi @joopy, Theo was so kind to accept my pull request #1951 and add it to 5.12.0b with commit ecfa14d28b4285a9022d508e6f5b27f7654bdbf5 already. It works fine for me. Maybe you can test this, too? Just make sure to comment out #define USE_TSL2561 and uncomment the #define USE_TSL2561_JOBA instead.

joopy, 2018-02-20:
Hi @joba-1 I can confirm it's working.

12:45:43 MQT: tele/sonoffTH16-2/SENSOR = {"Time":"2018-02-20T12:45:43","TSL2561":{"Illuminance":2623.772}}

arendst, 2018-02-20:
In that case I will remove the original driver and make jobas the standard one. Thnx for your support

joopy, 2018-03-01:
Is working fine a few weeks now can be closed.

closed 2018-03-01 by joopy