arendst/Tasmota #1646: sonoff touch 2 chanel mqtt state_topic:

vMeph, 2018-01-20:
is there anyway to add state_topic: to the sonoff touch?
im trying to control my sonoff 2 chanel with homeassistant
i can control on off with home assistant but then if i go turn off button with the sonoff the homeassitant will actualize the state is there anyway for this to be acomplish or iam missing some config?
current my HA config is :
light:
  - platform: mqtt
    name: "TEST"
    state_topic: "stat/sonoff/POWER"
    command_topic: "cmnd/sonoff/power2"
    qos: 0 
    payload_on: "ON" 
    payload_off: "OFF"
    optimistic: true

davidelang, 2018-01-20:
well, if you are sending a command for relay2 (power2), you should probably be 
looking for the state for POWER2 as well.

vMeph, 2018-01-20:
@davidelang thanks mate that was the problem

closed 2018-01-20 by vMeph