arendst/Tasmota #1647: the BME680 goes to sleep and read very low resistance when Tasmota web UI is not open

anpj123, 2018-01-20:
The “gas” readings from BME680 are correct only when the Sonoff Tasmota UI is open in browser. If it’s open then the reading is correct about 170 kOhm , but as soon as the browser is closed the reading reported via MQTT droops down to about 36 kOhm. The temperature and pressure are OK, but the GAS seems to not work. I think the sensor goes to sleep. Is there any way to keep it alive all the time or at least in some intervals?

arendst, 2018-01-20:
You might want to make the following change to file xsns_09_bmp.ino to see if it fixes the problem:
```
  if (i2c_flg) {
    switch (function) {
      case FUNC_PREP_BEFORE_TELEPERIOD:
        BmpDetect();
        break;
      case FUNC_EVERY_SECOND:
#ifdef USE_BME680
        if (tele_period == Settings.tele_period -3) {
          Bme680PerformReading();
        }
#endif  // USE_BME680
        break;
      case FUNC_JSON_APPEND:
        BmpShow(1);
        break;
 ```

anpj123, 2018-01-20:
I checked my 4 BME680 and they all need about 3-4min stabilization time to get a meaningful GAS reading. So I added the following code (below) to sample sensor every 1s for 5min before the telemetry reporting. With that modification firmware v 5.11.1b works perfect with different tele periods. 
Like always thanks for all your help!

  if (i2c_flg) {
    switch (function) {
      case FUNC_PREP_BEFORE_TELEPERIOD:
        BmpDetect();
      break;                                     // this line was added     
      case FUNC_EVERY_SECOND:                    // check every second 
        #ifdef USE_BME680
          if ((tele_period - Settings.tele_period) < 300) {     // reads BME680 300s before teleperiod - required for GAS stabilization
          Bme680PerformReading();
          BmpShow(1);                                         // added to append JSON 
          }                                                 // end if < 300s
        #endif  // USE_BME680
      break;                                    
   case FUNC_JSON_APPEND:
        BmpShow(1);
   break;

arendst, 2018-01-20:
In that case forget the FUNC_EVERY_SECOND part an add the Bmp680PerformReading function in the FUNC_JSON_APPEND part just below BmpShow(1).

anpj123, 2018-01-20:
When I put it in FUNC_JSON_APPEND then it doesn't work, as the sensor needs to be actively pulled at least every 2s (2s is in Adafruit example)  to stabilize. I see FUNC_JSON_APPEND is executed only when the telemetry is called. The current minimum for telemetry is 10s and that's too long.

pascalsaul, 2018-01-30:
This is the effect on mine BME680, is that as expected:
[image]

anpj123, 2018-01-31:
Did you try to keep the browser open with your Sonoff Tasmota firmware website ? Did you notice any spike during that time ? When I have the browser open my kOhms readings can double within 60s or so. When I close the browser then the readings goes down. I think this is the problem with the BME680 gas internal heater . In the library you can set the heater temperature and time with the flowing function:  
`bme.setGasHeater(320, 1000); // 320*C for 1000 ms  (Adafruit default is 150ms)`
I tested it on the regular Arduino  and I had to increase the heat time to around 1000ms to have reliable reading at the first pull request. If you pull the sensor frequently (like every 1s or so) and have low "heating time" then you can see that the sensor resistance gradually increase every time you pull the data. This is exactly what's happening when you have the Sonoff Tasmota device website open - the sensor is updating every 1s and your resistance increases. 
To transform the kOhms to  "IAQ Index" , I found the flowing post. This guy read resistance and made some conversions. On my sensors (I have 5 running now) I read about 300-400 kOhm in relatively fresh air and about 50 kOhms when I stick the sensor in the bag with my son's used diapers :-) 

> https://forums.pimoroni.com/t/bme680-observed-gas-ohms-readings/6608

pascalsaul, 2018-01-31:
You're right... as soon as I opened the webgui the kOhm increased immediately! It's almost doubled now in the browser...
[image]

closed 2018-06-15 by stale[bot]