arendst/Tasmota #1649: Sonoffs touch Restart and change power state

hwarang098tkd, 2018-01-20:
Hi,
I have 12 touch sonoffs for lamps . I use mqtt protocol. I notices that every 1 h all of them restart. But this is not the problem. One of them after the restart change from Off to On for no reason. I set Poweronstate to 3. 
Any solution?

ornolio, 2018-01-20:
me too: 2 out of 18 sonoff touch switch at mqtt reboot

Frogmore42, 2018-01-20:
This is likely not an easy problem to solve without any information.

What version of Tasmota are you using?
How do you have things configured?
Was it working before?

Pretty sure Tasmota will reboot if it loses connection with either Wi-Fi or MQTT.  What do your router logs and MQTT logs say?
What is the uptime listed in the console?

hwarang098tkd, 2018-01-21:
10:57:41 MQT: stat/sonoff_t_10/STATUS = {"Status":{"Module":10,"FriendlyName":"Sonoff","Topic":"sonoff_t_10","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:57:41 MQT: stat/sonoff_t_10/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","Uptime":27,"Sleep":0,"BootCount":218,"SaveCount":2500,"SaveAddress":"FA000"}}
10:57:41 MQT: stat/sonoff_t_10/STATUS2 = {"StatusFWR":{"Version":"5.11.1","BuildDateTime":"2018-01-07T21:51:09","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
10:57:41 MQT: stat/sonoff_t_10/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"David","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"55800009"}}
10:57:42 MQT: stat/sonoff_t_10/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
10:57:42 MQT: stat/sonoff_t_10/STATUS5 = {"StatusNET":{"Hostname":"sonoff_t_10-0034","IPAddress":"192.168.1.7","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"60:01:94:85:40:22","Webserver":2,"WifiConfig":3}}
10:57:42 MQT: stat/sonoff_t_10/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.3","MqttPort":1883,"MqttClientMask":"touch_10_bedroom_1","MqttClient":"touch_10_bedroom_1","MqttUser":"DVES_USER","MAX_PACKET_SIZE":512,"KEEPALIVE":15}}
10:57:42 MQT: stat/sonoff_t_10/STATUS7 = {"StatusTIM":{"UTC":"Sun Jan 21 09:57:42 2018","Local":"Sun Jan 21 10:57:42 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
10:57:43 MQT: stat/sonoff_t_10/STATUS10 = {"StatusSNS":{"Time":"2018-01-21T10:57:42"}}
10:57:43 MQT: stat/sonoff_t_10/STATUS11 = {"StatusSTS":{"Time":"2018-01-21T10:57:43","Uptime":27,"Vcc":3.123,"POWER":"OFF","Wifi":{"AP":1,"SSId":"David","RSSI":78,"APMac":"A4:B1:E9:9C:17:B0"}}}

I cant find router logs, mqtt log says that sonoff go offline and then online, openhab says that the state changed from OFF to ON after restart.

davidelang, 2018-01-21:
set logging to 'more detail', but also see the wiki page on poweronstate and 
retain. This really looks like your mqtt server is retaining messages that get 
sent to your device when it connects.

Frogmore42, 2018-01-21:
The log indicates that this one has been up for 27 hours, so it is not rebooting every hour, but it probably is reconnecting to MQTT.  Looking at the log for a longer period of time would show that and David's assessment is probably correct too.

hwarang098tkd, 2018-01-23:
ok i searched it and run this through cmd 'mosquitto_sub -p 1883 -t cmnd/sonoff_b1_1/power'and the result was : 'ON'
I guess that this is a retain message from this topic and will sent the command 'ON'
who to avoid this? or delete it every time

davidelang, 2018-01-23:
find out what's sending the command with the retain flag set and change it.

what are you using for your home automation software?

hwarang098tkd, 2018-01-23:
openhab...

ornolio, 2018-01-24:
MQTT DASH APP

 

Thank You

 

Pier Giacomo Ornolio
mobile: +39.335.7831.433
ornolio@gmail.com or <mailto:ornolio@gmail.com%20or>  ornolio@libero.it
posta certificata: ing.piergiacomo.ornolio@pec.it
www.skype.com 'pgornolio'

  _____  

A: arendst/Sonoff-Tasmota
Cc: ornolio; Comment
state (#1649)

 

 

 

<https://github.com/arendst/Sonoff-Tasmota/issues/1649#issuecomment-35993381
8>  it on GitHub, or mute
ft7Gf5aks5tNkyngaJpZM4RlVOt>  the thread.
<https://github.com/notifications/beacon/ADR9v8hWZpRf4ksnRmGLGKYtb8D0VIhLks5
tNkyngaJpZM4RlVOt.gif>

closed 2018-06-08 by stale[bot]