arendst/Tasmota #1652: mqtt INFO2 publish

rene9900, 2018-01-20:
is there some kind of way to change the publish rate of tele/sonoff/INFO2? as it is now it seems like it only publish the ip at restart og the tasmota device, so most often i have my ip field blank after several weeks uptime :(
[image]

arendst, 2018-01-20:
Great about the uptime.

The Info lines are programmed to only show on reboot, sorry. You might want to trigger the ``status 5`` message which contains the IP address too.

rene9900, 2018-01-20:
great thanks. 
status 5 command, but how do i make the tasmota firmware do a "status 5" with the teleperiod ? or some kind of interval besides manually or restarting?

arendst, 2018-01-20:
You can't!!

Need external trigger like a python script executing a mqtt command to the GroupTopic

rene9900, 2018-01-20:
ok nice, i didn't think about group topic since i never saw a use for it until now 👍 
python or node-red? i made a node in node red to trigger all my sonoffs which all have "sonoffs" as group topic, but nothing appears in the console of the module, i have set it to 10 sec interval to test, what would the correct command be to hit my sonoffs?

[image]

arendst, 2018-01-20:
I don't know your fulltopic setup but I would expect to see a prefix like ``cmnd``

rene9900, 2018-01-20:
ah iforgot that one, im going to try some combinations of the commands, my cmnd/sonoffs : msg.payload : string[8]
"status 5"
responds on all my devices but with command unknown, so it has to be some minor error

16:56:28 MQT: stat/miniD1/COMMAND = {"Command":"Unknown"}
16:56:38 MQT: stat/miniD1/COMMAND = {"Command":"Unknown"}
16:56:48 MQT: stat/miniD1/COMMAND = {"Command":"Unknown"}
16:56:58 MQT: stat/miniD1/COMMAND = {"Command":"Unknown"}

rene9900, 2018-01-20:
Problem solved, thank you for helping me on the right path..
[image]

closed 2018-01-20 by arendst