arendst/Tasmota #1653: Sonoff Basic - 2 external switches with different switchtopic possible?

sfanicola, 2018-01-20:
Sorry if the question was made already but I found more than one issues post about confusing me more and more.
I've got a sonoff basic with 2 external switches attached at GPIO 3 and 14, one set as Switch1 and the other as Switch2.
Is it possible to set the switch topic just for the second switch so that it controls another sonoff in my lan via mqtt?

davidelang, 2018-01-20:
no,switchtopic applies to all switches

b1g1an, 2018-01-21:
Apologies if this is a dumb way of doing it but I was totally confused how to do this as well but eventually achieved the desired result like this... in the web config set GPIO03 as Relay2, which gives me another toggle in the web ui home page, and set GPIO04 to Switch2.

Switch1 on GPIO14 still controls the Sonoff directly as before, so works if wifi is down, but the MQTT message the Sonoff generates has now gained a '1' on the end, ie 'stat/sonoff2/POWER1', so have amended the switch in Home Assistant to reflect this. A second switch connected between GPIO04 and Gnd now gives stat/sonoff2/POWER2' so in Home Assistant I just use a change in that state to trigger another Sonoff although obviously it could be used to trigger anything.

If there is a better way to achieve this I'm all ears?

sfanicola, 2018-01-21:
my question was about using one sonoff to control 2 different sonoffs, and without home assistant similar problem this is not possible.
I may be wrong but you don't need setting the basic as dual, but just set switchtopic1 and switchtopic2

davidelang, 2018-01-21:
There is only one switchtopic, not one per switch.

b1g1an, 2018-01-21:
@davidelang  any opinion on my solution for my situation?

davidelang, 2018-01-21:
set the switchtopic and have your home automation software respond to it and 
send the commands out to different other things.

b1g1an, 2018-01-22:
Thanks, must've misunderstood what I'd read as I'd got the impression setting a switchtopic stopped the direct action of switch 1, instead relying on the MQTT server running for operation.

davidelang, 2018-01-22:
On Mon, 22 Jan 2018, b1g1an wrote:

> Thanks, must've misunderstood what I'd read as I'd got the impression setting 
> a switchtopic stopped the direct action of switch 1, instead relying on the 
> MQTT server running for operation.

Yes, setting switchtopic means that all switch actions go through the mqtt 
server.

iandronowicz, 2018-05-22:
So the solution explained by @b1g1an still relays on mqtt server to turn on the wired light?

I’m setting SwitchTopic for the same reason: one switch to toggle the wired light and the other for triggering a rule in openhab2. Don’t set a relay2 just switch1 and switch2. The downside is that if the mqtt server is unreachable for some reason the wired light won’t turn on.

Have any of you found a better way to do this?

ascillato, 2018-05-22:
If you are using last Tasmota version 5.14.0a

**Use rule for the switch1:**

_Turn on rules_
on console: `backlog rule 1; rule 4`

_Add rule_
on console: `rule on switch1#state do power1 on endon`

ascillato, 2018-05-22:
The command run on the rule should be `power1 2` (toggle) if you have `switchmode1 3`.

See wiki for more information on switchmode

iandronowicz, 2018-05-22:
@ascillato thanks for the tip on rules... didn’t even know they existed. And yes, I am running the latest tasmota version.

I will look further into this... didn’t quite understand it

iandronowicz, 2018-05-23:
@ascillato do you know how to delete all rules? thanks!

iandronowicz, 2018-05-23:
I can't seem to get rid of the rule:

01:26:48 CMD: on switch2#state do publish cmnd/sonoffbasic8/power2 %value% endon
01:26:48 MQT: stat/sonoffbasic8/RESULT = {"Command":"Unknown"}
01:26:53 CMD: rule
01:26:53 MQT: stat/sonoffbasic8/RESULT = {"Rule":"OFF","Once":"OFF","Rules":"on switch1#state do power1 2 endon"}

I want to get rid of using switchtopic, and set the 2nd external switch to publish a mqtt message to the server and, of course, not to change the state of the relay

ascillato, 2018-05-23:
Commands on console:

Rule 0
Rule none

With that you erase all rules.

ascillato, 2018-05-23:
I think that in your publish command power1 should be POWER1

ascillato, 2018-05-23:
To save a rule you have to use the rule command as:

Rule on switch......

On switch.... Will not work

iandronowicz, 2018-05-23:
thanks... that was the problem!

Im going nuts though... The rule:

rule on switch2#state do publish cmnd/sonoffbasic8tecla2/POWER %value% endon

works like a charm, the mqtt server gets the message and everything works but... I don't know why when I use the external switch2 it ALSO toggles the relay...

iandronowicz, 2018-05-23:
The switch1 is connected to GPIO3 and the switch2 to GPIO14

iandronowicz, 2018-05-23:
Im starting to think that it may be a wiring problem...

chrishiscox, 2018-07-08:
I am having issues just enabling the rules engine. I have issued (in user_config.h):

#ifndef _USER_CONFIG_H_
#define _USER_CONFIG_H_
#define USE_RULES
I compile, then OTA to the Sonoff 4ch Pro.
When I type rule into CMD I get command not recoginised. 
I am using Tasmota 5.12

Any ideas?

Chris

ascillato2, 2018-07-10:
Hi,

@chrishiscox Latest release have rules. 5.12 does not have rules feature implemented.

@sfanicola Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks.

closed 2018-07-10 by ascillato2