arendst/Tasmota #1654: Version v5.11.1d  Compiler Error

micropet, 2018-01-20:
v5.11.1d gives the following compiler error with Arduino:

D:\arduino-1.8.3\meine\Sonoff-Tasmota\sonoff\webserver.ino: In function 'void ShowPage(String&, bool)':

webserver:850:45: error: default argument given for parameter 2 of 'void ShowPage(String&, bool)' [-fpermissive]

   String page = FPSTR(HTTP_HEAD);
                                         ^

webserver:425:6: error: after previous specification in 'void ShowPage(String&, bool)' [-fpermissive]

 void ShowPage(String &page, bool auth = true)

      ^

exit status 1

arendst, 2018-01-20:
I've just tried compiling with Arduino IDE 1.8.1 and 1.8.5 and they both run fine. I have no 1.8.3 available to test.

I've seen such problems before and I propose to make the following change in file webserver.ino at line 425:
```
void ShowPage(String &page, bool auth)
```
so removing the assignment to auth.

Add the following code starting at line 451:
```
void ShowPage(String &page)
{
  ShowPage(page, true);
}
```
This works for all versions and compilers.

micropet, 2018-01-20:
OK Theo, works again.

I use version 1.8.5 or 1.9 Beta. 
Testet with 1.9 Beta
Thank you.

Greetings Peter

closed 2018-01-20 by arendst