arendst/Tasmota #1655: Compilation error. 'XdrvSetPower' was not declared in this scope

soswow, 2018-01-20:
I think I've followed all the instructions. Getting this on first compilation try:

```
/sonoff/sonoff.ino: In function 'void SetDevicePower(power_t)':
sonoff:308: error: 'XdrvSetPower' was not declared in this scope
   XdrvSetPower(bitRead(rpower, devices_present -1));
...
...
```

arendst, 2018-01-20:
Pls download the latest version. This function has been changed recently and you seem to have an new version sonoff.ino and an old version xdrv_interface.ino

soswow, 2018-01-20:
Sure. Just for the reference. I've downloads latest zip from releases page
couple hours before submitting a ticket. I didn't touch source code much.

soswow, 2018-01-20:
My Arduino sketch folder: `~/Documents/Arduino` I downloaded https://github.com/arendst/Sonoff-Tasmota/archive/v5.11.1.zip Unzipped. Moved all the folders from `/lib` to `~/Documents/Arduino/libraries`, and copied `/sonoff` folder into `~/Documents/Arduino`. Opened Arduino IDE, "Open" in menu and opened `~/Documents/Arduino/sonoff/sonoff.ino`. Configured Board to be 8266 one with settings defined here https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE#configure-arduino-ide
Get this error.

arendst, 2018-01-21:
All procedures look OK. 

I can understand why Arduino complaints about undeclared functions but then I cannot understand why it works with my Arduino IDE's both 1.8.1 and 1.8.5.

What version of Arduino IDE are you using?

soswow, 2018-01-23:
I am still on 1.6.9 Will upgrade and try again

soswow, 2018-01-23:
It solved a problem. Thank you!

closed 2018-01-23 by soswow

PMAHULK13, 2018-06-05:
I have followed these instructions and still receiving this error. any ideas?

chrkoe, 2018-07-06:
I got the same error and got here by googling it.
Arduino IDE 1.8.5
Ubuntu 18
Sonoff 4CH
My error message(s) are very long. I add it here, maybe it will help?!
[compile-error.txt](https://github.com/arendst/Sonoff-Tasmota/files/2170886/compile-error.txt)