arendst/Tasmota #1656: UPP: WeMo Type 2 UPP to NAS

zibous, 2018-01-20:
I do not understand why the program version 5.11.1d UPP sent to one of my NAS devices? Should not this be the Amazon Alexa device?

Can I configure this?

```
16:28:19 HTP: WeMo basic event
16:28:19 MQT: stat/M1120N/RESULT = {"POWER":"OFF"}
16:28:19 MQT: stat/M1120N/POWER = OFF (retained)

16:28:22 UPP: WeMo Type 2, Response sent to 10.1.1.146:38317

16:28:37 WIF: Checking connection...
16:28:37 WIF: Connected
16:28:46 HTP: WeMo basic event
16:28:46 MQT: stat/M1120N/RESULT = {"POWER":"ON"}
16:28:46 MQT: stat/M1120N/POWER = ON (retained)
16:28:46 CFG: Saved to flash at F6, Count 425, Bytes 1648
```

zibous, 2018-01-20:
```
Starting Nmap 7.40 ( https://nmap.org ) at 2018-01-20 16:17 CET
Nmap scan report for WDNAS2015.local (10.1.1.146)
Host is up (0.016s latency).
Not shown: 987 closed ports
PORT      STATE SERVICE
21/tcp    open  ftp
22/tcp    open  ssh
80/tcp    open  http
111/tcp   open  rpcbind
139/tcp   open  netbios-ssn
443/tcp   open  https
445/tcp   open  microsoft-ds
548/tcp   open  afp
2049/tcp  open  nfs
3689/tcp  open  rendezvous
8181/tcp  open  intermapper
9000/tcp  open  cslistener
49152/tcp open  unknown
```

arendst, 2018-01-20:
uPnP M-SEARCH is send by many devices in (my) network like Kodi, Sonos, Router etc. Sonoff just responds to the default M-Search message. I still have no Alexa to test...

In previous version it listened to a specific Wemo M-Search request but people complaint about their Alexa version 2  was unable to see Sonoff emulating. After responding to the generic M-Search they were happy again.

zibous, 2018-01-20:
Hi Theo 

I use Echo Plus, Echo Version 2 and Dot (2nd generation). Alexa is not always able to find the Sonoff (WeMos) devices, but turns them on or off.

That's why I was surprised that the uPnP M-SEARCH was not sent to an echo or dot device.

But the error could also be due to the Amazon Echo / Dot software.
But maybe that will change again ...

arendst, 2018-01-21:
The M-SEARCH should originate from the Echo as that device is looking for hardware to manage. The Hardware then needs to respond with it's IP address. Tasmota will never send uPnP on it's own. It just listens for M-SEARCH messages.

zibous, 2018-01-21:
Hi Theo,

Thank you, understand now - so when other devices makes a uPnP request, tasmota listens to it and answers the request.

closed 2018-01-21 by zibous