arendst/Tasmota #1659: RF-Bridge no Update possible

Beliar-666, 2018-01-20:
Hello, my RF Bridge running 5.10 will not Update. Everytime i try via File upload, with the minimal file it says `Sonoff Bridge Module
Sonoff
Upload Failed

Not enough space`

Btw. i wondered why the Bridge has no IDX Fields for the 16 RF Channels?

arendst, 2018-01-21:
Report your free space. It should be around 488kB. Small enough to load the minimal version. Unless you initially flashed the device with the wrong Flash config like 1M and any SPIFFS.... which should have been 1M (No SPIFFS).

Beliar-666, 2018-01-21:
Hi.

`
ESP Chip Id | 9329635
Flash Chip Id | 1327185
Flash Size | 1024kB
Program Flash Size | 1024kB
Program Size | 486kB
Free Program Space | 4kB
Free Memory	23kB
`

I flashed it like my T1 Dual. The Dual had no issues updating to 5.11. So i wondered why the Bridge wont. 

I just took a look to my T1 and saw:

`
Free Program Space | 516kB
`

Now is there any way to fix it?

arendst, 2018-01-21:
That looks like a flash size of 512kB to me. Totally wrong. You need 1M (NoSpiffs).

Use the correct flash config. Only solution now is connect serial and re-flash.

Beliar-666, 2018-01-21:
this is what i tried since days. it wont.

Arduino IDE always shows:

`warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed`

Joeyhza, 2018-01-21:
Download this

NodeMCU-PyFlasher-2.1

It's the only tool i use for flashing and all firmware files to all my devices

In the tool make sure you select DOUT and the option to erase the current flash before it writes the new one

Beliar-666, 2018-01-21:
Thanks to all. Update worked for me. 

But there is also the Problem with the IDX for the 16 Switches.....

closed 2018-06-05 by stale[bot]