arendst/Tasmota #1661: MagicHome LED  (aka Flux-Led) - can't flash tasmota until end

HarryHase11, 2018-01-21:
I have problem to falsh this controller, after a shot time flashing faild without ending.
I killed one controller.
With the Second controller, if it stops i flashed the original bootloader, then i can start new.
-
I used normal tasmota (8285 - 5.10.0) as on all other (20) devices. What i have to change?

Jason2866, 2018-01-21:
Hi,
tried to flash with esptool with option DOUT? Erase the flash before! You can use esptool for that too.
I recovered a dead NodeMCU (my testting unit for new Tasmota things) this way...

HarryHase11, 2018-01-21:
Erease means to write a blank 1MB to the controller?

Jason2866, 2018-01-21:
No,
esptool.py --port COMx --baud 115200 erase_flash

a full description (Linux) for Backup Writing is analog...

Backup stock firmware
This procedure is provided AS-IS, just like the rest of this project, this has worked for me but might not work for you. It's a simple dump of the flash contents, including emulated EEPROM data and unaware of the flash layout. So it's meant for simple backup & restore procedure.

Requirements
You will need a working copy of esptool.py. You might already have it with you IDE of choice, make sure you locate it and it's up to date.

The examples below assume you have esptool.py somewhere in the path, otherwise specify the full path to the tool.

Connection
To backup the image you will need to boot your board in flash mode. The procedure is exactly the same as when you are getting ready to flash a new image and it will depend on your device. Check the supported hardware page for instructions.

Backup
Depending on your device flash memory size you will need to specify a different size. Check the read_flash options:

$ esptool.py read_flash -h
usage: esptool read_flash [-h] [--progress] address size filename

positional arguments:
  address         Start address
  size            Size of region to dump
  filename        Name of binary dump

optional arguments:
  -h, --help      show this help message and exit
  --progress, -p  Show progression
To backup a Sonoff device (1Mbyte flash size) you will need to (change port to match yours):

esptool.py --port /dev/ttyUSB0 --baud 115200 read_flash --progress 0x00000 0x100000 sonoff-backup.bin
To restore the image do a:

esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash -fs 8m -fm dio 0x00000 sonoff-backup.bin
The writing process first erases the flash contents. In case you wanted to do it manually you can:

esptool.py --port /dev/ttyUSB0 --baud 115200 erase_flash

baddle, 2018-01-21:
I also had problems flashing my magic-home-controller. 
I also thought in between that I killed him.
For me the solution was to flash sonoff-minimal first and after this I flashed the complete firmware over the web-gui.

closed 2018-06-04 by stale[bot]