arendst/Tasmota #1662: Upgrade issue with the latest version v5.11.1d

micropet, 2018-01-21:
I'm currently flashing a Wemos D1 and a NodeMCU.
(Dout 1M no SPIFFS).
Both work.

If I want to flash for an test update the minimal version with "Upgrade by file upload" I get the error message:

Upload Failed
Not enough space

Here is the information from the NodeMCU:

ESP Chip Id 14694793
Flash Chip Id 1458400
Flash Size 4096kB
Program Flash Size 1024kB
Program Size 512kB
Free Program Space 488kB
Free Memory 24kB

Peter

arendst, 2018-01-21:
How do you compile your minimal version? Mine has a size of approx. 451kB which fits nicely in your 488kB free Program Space...

emontnemery, 2018-01-21:
OTA upgrade also fails for me with latest version, but with error 5 (Upload buffer miscompare).
firmware.bin is 519kB, is that too much to support OTA with 1MB flash?

arendst, 2018-01-21:
You should know that OTA needs FREE PROGRAM SPACE to load the new firmware. Once loaded a reboot will overwrite the original firmware by copying the new firmware from former FREE PROGRAM SPACE to it's working location.

You see both errors (Upload buffer miscompare or Not enough Space) as the file you want to upload is bigger than the amount of FREE PROGRAM SPACE.

So to get your required program loaded you will have to load a program first which is smaller than FREE PROGRAM SPACE. That happens to be the minimal version as that's what I have designed it for. Once the minimal version is installed you'll see a much bigger FREE PROGRAM SPACE which is large enough to load your required program. So now upload your required program and you're done.

So a firmware.bin file of 519kB is no problem for OTA as long as you perform the two step approach.

See https://github.com/arendst/Sonoff-Tasmota/wiki/What's-New where I've tried to explain the same...

micropet, 2018-01-21:
Hi Theo,

i compile with
#define BE_MINIMAL
and its 461529 big.

micropet, 2018-01-21:
Ich found a minimal File with 434 k.
But i get:

Upload Failed
Not enough space

arendst, 2018-01-21:
@emontnemery that's the consequence of using 2.4.0. There is a discussion going on there (https://github.com/esp8266/Arduino/issues/4089 and https://github.com/esp8266/Arduino/issues/4172) about including these features where some say if you don't use it it will not be in the image but as you see it's there and I definitively do not use them as I had to write my own small functions to get this functionality in 2.3.0.

I hate it too and expect someone over there will someday see the light and get rid of it.

@micropet I'm stunnend for the moment.

pascalsaul, 2018-01-21:
I had also problems with .bin' over ~490kb... Suddenly al my devices are getting disconnected every 5 minutes :/ So my domotics was very instable :(

emontnemery, 2018-01-21:
@arendst OK, I see. As a workaround, would it be possible to replace the unneeded float parsing/formatting functions with empty stubs? Maybe it would require an additional build step where the unnecessary symbols are first removed from libc.a, or include a custom libc.a with tasmota without floating point string formatting/parsing support.

micropet, 2018-01-21:
Oh crap.

Then OTO would never work again?
But something is not right:

Why does a 434kB update not fit into the ESP 488kB Free Program Space?

Peter

martin072, 2018-05-20:
Having the same issues, found a relative simpel work around, as #define BE_MINIMAL didn't do the job for me either;

- Compile the Arduino Example ESP8266 HTTP WebUpdater.
- Upload it to the Tasmota device
- Browse to: esp8266-webupdate.local/update
- Upload the Tasmota binary
Done.

ascillato, 2018-06-13:
Hi

If your issue was solved, please close it. Thanks :+1:

closed 2018-06-13 by ascillato2

riddik14, 2020-09-09:
versio 8.5.0 
from consolle
from ota or manual and setoption78 1
stat/UPGRADE = {"Upgrade":"Failed Not Enough space"} with minimal version... why?

corbolais, 2021-07-05:
> Having the same issues, found a relative simpel work around, as #define BE_MINIMAL didn't do the job for me either;
> 
> * Compile the Arduino Example ESP8266 HTTP WebUpdater.
> * Upload it to the Tasmota device
> * Browse to: esp8266-webupdate.local/update
> * Upload the Tasmota binary
>   Done.

Situation:
* tasmota-minimal v9.5.0.2 already flashed.
* Upgrading using other tasmota-minimals fails due to "not enough space".
* Upgrading by the path you suggested using Webupdater is failing as well.
* WebUpdater binary is ~287kB whereas tasmota-minimal is ~270kB.

ascillato, 2021-07-05:
This is a very old closed issue. Please, address your issue in the chat please.