arendst/Tasmota #1663: IR Receive, always receive same code

Jason2866, 2018-01-21:
Hello,
when using IR Receive with different remotes and buttons i always get the same received code:
13:17:42 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:42 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:45 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:46 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:47 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:52 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:53 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:55 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:58 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:18:00 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:18:01 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:18:03 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:18:05 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
I think thats not the way it should work?
Thanks for helping in advance!
Greets J.Obermeier

cinehorse, 2018-01-21:
I have the same problem, all keys have the same "Data":
```
22:48:59 MQT: tasmota/wemosd1/tele/RESULT = {"IR empfangen":{"PROTOKOLL":"RC6","BITS":20,"DATEN":"0"}}
22:48:59 MQT: tasmota/wemosd1/tele/RESULT = {"IR empfangen":{"PROTOKOLL":"RC6","BITS":20,"DATEN":"0"}}
22:49:00 MQT: tasmota/wemosd1/tele/RESULT = {"IR empfangen":{"PROTOKOLL":"RC6","BITS":20,"DATEN":"0"}}
22:49:00 MQT: tasmota/wemosd1/tele/RESULT = {"IR empfangen":{"PROTOKOLL":"RC6","BITS":20,"DATEN":"0"}}
22:49:01 MQT: tasmota/wemosd1/tele/RESULT = {"IR empfangen":{"PROTOKOLL":"RC6","BITS":20,"DATEN":"0"}}
22:49:02 MQT: tasmota/wemosd1/tele/RESULT = {"IR empfangen":{"PROTOKOLL":"RC6","BITS":20,"DATEN":"0"}}
22:49:02 MQT: tasmota/wemosd1/tele/RESULT = {"IR empfangen":{"PROTOKOLL":"RC6","BITS":20,"DATEN":"0"}}
22:49:03 MQT: tasmota/wemosd1/tele/RESULT = {"IR empfangen":{"PROTOKOLL":"RC6","BITS":20,"DATEN":"0"}}
```
RC6 is correct, is a Philips Remotecontrol from a 42PFL8404 (with Ambilight)
so I  can't found a similar RC on LIRC Homepage to verfied the "Bits"

mhamon5, 2018-01-22:
I'm in a similar situation. Different remotes are detected with different protocols, but no Data is presented.

{"IrReceived":{"PROTOCOL":"SAMSUNG","BITS":32,"DATA":"0"}}
{"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"0"}}

Using a Sonoff DEV board - Tasmota compiled using platformio.

brianbrown1962, 2018-01-22:
Same here...always receiving DATA "0" in console regardless of type of remote and/or keys pressed.
.../tele/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"0"}}

Using a NodeMCU v1.0 with Tasmota 5.11.1

rewegit, 2018-01-27:
Same at me.
- V5.11.1g
Arilux LC03 (modified LC01)
00:35:42 MQT: RGB1/tele/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"0"}}

rewegit, 2018-01-29:
The problem seems to be snprintf_P in line 110 in xdrv_02_irremote. ino.
ESP8266 snprintf probably doesn't handle 64bit values. 
For protocols with 32bit the following workaround works for me:

`
xdrv_02_irremote.ino
Line 110:

snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("{\"" D_JSON_IRRECEIVED "\":{\"" D_JSON_IR_PROTOCOL "\":\"%s\",\"" D_JSON_IR_BITS "\":%d,\"" D_JSON_IR_DATA "\":\"%lX\"}}"),
        GetTextIndexed(sirtype, sizeof(sirtype), iridx, kIrRemoteProtocols), results.bits, (uint32_t) results.value);
    
MQTT Example:
16:51:26 MQT: RGB1/tele/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FFF807"}}
16:51:31 MQT: RGB1/tele/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FFB04F"}} 
`

johoberm, 2018-01-29:
I get error message(s) in Platformio:
Compiling .pioenvs\sonoff\src\sonoff.ino.o
In file included from C:\users\hans\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:240:0,
from c:\users\hans\appdata\local\temp\tmprwbddn:1:
D:/Sonoff-Tasmota-master/sonoff/xdrv_02_irremote.ino: In function 'void IrReceiveCheck()':

D:/Sonoff-Tasmota-master/sonoff/xdrv_02_irremote.ino:113:109: error: 's' was not declared in this scope
snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("{"" D_JSON_IRRECEIVED "":{"" D_JSON_IR_PROTOCOL "":"%s","" D_JSON_IR_BITS "":%d,"" D_JSON_IR_DATA "":"%lX"}}"),
^
C:\users\hans\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/pgmspace.h:21:68: note: in definition of macro 'PSTR'
#define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
^
D:/Sonoff-Tasmota-master/sonoff/xdrv_02_irremote.ino:113:110: error: expected ')' before string constant
snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("{"" D_JSON_IRRECEIVED "":{"" D_JSON_IR_PROTOCOL "":"%s","" D_JSON_IR_BITS "":%d,"" D_JSON_IR_DATA "":"%lX"}}"),
^
C:\users\hans\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/pgmspace.h:21:68: note: in definition of macro 'PSTR'
#define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
^
*** [.pioenvs\sonoff\src\sonoff.ino.o] Error 1

johoberm, 2018-01-29:
This one works for me:

        snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("{\"" D_JSON_IRRECEIVED "\":{\"" D_JSON_IR_PROTOCOL "\":\"%s\",\"" D_JSON_IR_BITS "\":%d,\"" D_JSON_IR_DATA "\":\"%lX\"}}"),
        GetTextIndexed(sirtype, sizeof(sirtype), iridx, kIrRemoteProtocols), results.bits, (uint32_t) results.value);

Thanxxx a lot rewegit!!!

johoberm, 2018-01-29:
Console Log:

19:22:11 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FF609F"}}
19:22:13 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FF906F"}}
19:22:15 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FF906F"}}
19:23:13 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FF22DD"}}
19:23:14 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FFF807"}}
19:23:17 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FF52AD"}}
19:23:19 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FF12ED"}}
19:23:20 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FF10EF"}}
19:23:37 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FFC837"}}
19:23:38 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FFE817"}}
19:23:46 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FF0AF5"}}
19:23:47 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FF08F7"}}
19:23:49 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FF6A95"}}
19:23:50 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FF2AD5"}}

arendst, 2018-02-03:
In next pre-release

closed 2018-02-03 by arendst

JackNJack, 2019-07-19:
Can we reopen this please? I have Hitachi ACU remote control that receives the same code for any keys/button that i press.
`08:01:36 RSL: RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":50,"Data":"0x00000420B6338"}}
08:01:40 RSL: RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":50,"Data":"0x00000420B6338"}}
08:01:44 RSL: RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":50,"Data":"0x00000420B6338"}}
08:01:46 RSL: RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":50,"Data":"0x00000420B6338"}}
08:01:48 RSL: RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":50,"Data":"0x00000420B6338"}}
08:01:50 RSL: RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":50,"Data":"0x00000420B6338"}}
08:01:52 RSL: RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":8,"Data":"0x844108B6"}}
08:01:54 RSL: RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":4,"Data":"0x6819A1AA"}}
08:01:55 RSL: RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":50,"Data":"0x00000420B6338"}}
08:01:57 RSL: RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":50,"Data":"0x00000420B6338"}}`

That is the log for the 9 buttons, sometimes some appears 12bits some 8 or 4.
Thank you for your help!

ascillato, 2019-07-19:
Hi,

This is a very old fixed issue.

Please, address your problem to the Tasmota Support Chat and if can't be solved with their help, you should open a new issue with all the information of template completed.

Thanks