arendst/Tasmota #1664: Difficult to connect to wifi / reboots / power cycle

myozone, 2018-01-21:
I don't think I'm the only one to have this error looking at Google - but how the heck do get it to work ?

My board is WAVGAT D1 mini with an ESP8266MOD from [here ](https://www.aliexpress.com/item/D1-mini-Mini-NodeMcu-4M-bytes-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266-by/32662942091.html?spm=a2g0s.9042311.0.0.Y7cggZ)

I've tried changing virtually every  Arduino tool upload setting but now I'm at a loss to know what to do. 

The board only appears to connect when I press the RESET button then it may connect, power disconnect/reconnect it just wont connect to my router ?

This the output from the serial port 

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_AC7DB4, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Connecting to AP2 NOSEY in mode 11N as sonoff-7604...
00:00:15 WIF: Connect failed with AP timeout
00:00:15 WIF: Connecting to AP1 NOSEY in mode 11N as sonoff-7604...
00:00:30 WIF: Connect failed with AP timeout
00:00:30 WIF: WPSConfigactive for 3 minutes
00:00:49 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

Dave

Frogmore42, 2018-01-21:
How are you powering it?
Some computer USB ports do not supply enough power to fully power an esp8266.  If yours is one of those a powered USB hub would likely help (the USB3 ones tend to be more capable in supplying power and work fine with the USB2 serial controller).

What router/AP are you using?
What security settings?

Have you tried connecting to the Wi-Fi network the esp8266 makes when it can't connect and see what it see?

myozone, 2018-01-21:
Hello, I thought the thing about power, it's powered from a Dell laptop (USB2) also tried a Samsung power supply. The router is an Draytek 2600VG. It seems to connect but very randomly - not sure way it makes a difference but when I press the reset button then it'll connect most of the time...

Frogmore42, 2018-01-22:
The esp8266 draws very large peak currents, well above the spec for USB 2.0 (which only specifies 500mA, though many hosts support more).  Many times it will work just fine, so this might not be your problem.  But, since you are having a problem, eliminating all the things it might be helps find the (one or few) thing(s) it is.

You might have a router that is slightly off spec in some way, which then sometimes works and other times doesn't.  I had not heard of Draytek before.  The model you have seems to be very old, so you might want to consider getting a new one.  Network connection issues are quite hard to troubleshoot when you have the right equipment, without it you have to replace stuff until you find something that works.  Have you tried more than one esp8266?  I have one that is really flakey.  If it was the only one I had, I would have been pulling my hair out.  Luckily, I had several and tried a different one, and it just worked with exactly the same setup and software.  I have no idea why one of them is so flakey, since they are so inexpensive it is not worth my time to figure out why.  Maybe one day when I am really, really, really bored I will try and figure it out 😀

myozone, 2018-01-22:
I think I've eliminated the power issue, I tested with 2.4A x2 USB power supply with a very short 20AWG USB cable and the issue is still there. What I did thwn was set up another router slightly newer ;) than my Draytek and it appears to connect very quickly. 

Another issue I have is uploading a new sketch doesn't always work. How I found this out is when I put in the SSID and Password for the 'new' router it didn't update so I updated via 192.168.4.1, it could be the Arduino settings I have ?

Frogmore42, 2018-01-27:
How are you uploading the sketch?
It sounds more like you did not update the config number in settings.  It is what determines if the code uses the values saved or the new default values.

myozone, 2018-01-27:
I used the  Arduino IDE, I'm not sure what mean by number ?

Frogmore42, 2018-02-03:
define CFG_HOLDER             0x20161209

Which is in user_config.h

closed 2018-06-18 by stale[bot]