arendst/Tasmota #1665: Sonoff T1 3CH UK constant reboots and resets back to Sonoff Basic

Creamola, 2018-01-21:
Uploaded 5.8.0 (full) via Atom over serial.

This is probably my mistake:

OTA to 5.11.1 - result was that it appeared to be dead.

Re-uploaded 5.8.0. That worked.

OTA to 5.11.1 minimal. That worked.

OTA to 5.11.1 full. Appeared to be dead.

Re-uploaded 5.8.0. That worked.

OTA to 5.9.1 - works, but constantly in a crash loop with via the serial:

epc1=0x402201cc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0):

Revert back to 5.8.0 - which is stable ONLY as Sonoff Basic. 

If I configure as T1 3CH, it reboots and comes back up as 3CH in the web interface, but almost immediately then reboots and reverts back to Sonoff Basic.

Console shows:

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_974377, GroupTopic sonoffs) Version 5.8.0
00:00:00 WIF: Connecting to AP1 SoX in mode 11N as sonoff-0887...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on sonoff-0887.local with IP address 192.168.0.128
00:00:04 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
00:00:04 RSL: stat/sonoff/POWER1 = ON
00:00:06 MQT: Attempting connection...
15:18:56 DNS: Query done. MQTT services found 1
15:18:56 DNS: MQTT service found on OpenHAB, IP Address 192.168.0.15, Port 1883
15:18:56 MQT: Connected
15:18:56 MQT: tele/sonoff/LWT = Online (retained)
15:18:56 MQT: cmnd/sonoff/POWER = 
15:18:56 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff T1 3CH", "Version":"5.8.0", "FallbackTopic":"DVES_974377", "GroupTopic":"sonoffs"}
15:18:56 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin", "Hostname":"sonoff-0887", "IPAddress":"192.168.0.128"}
15:18:56 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
15:18:57 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
15:18:57 MQT: stat/sonoff/POWER1 = ON
15:18:57 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
15:18:57 MQT: stat/sonoff/POWER2 = OFF
15:18:57 MQT: stat/sonoff/RESULT = {"POWER3":"OFF"}
15:18:57 MQT: stat/sonoff/POWER3 = OFF00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_974377, GroupTopic sonoffs) Version 5.8.0

Than it crashes and reboots as a Sonoff Basic again and shows restart reason as "Blocked loop":

Program Version	5.8.0
Build Date & Time	2018-01-21T12:25:24
Core/SDK Version	2_3_0/1.5.3(aec24ac9)
Uptime	0 Hours
Flash write Count	8 at F9000
Boot Count	2
Restart Reason	Blocked Loop
Friendly Name 1	Sonoff

Bottom line: I am cool with running this with 5.8.0, but it will only run as a Sonoff Basic - which isn't much use :-)

Any bright ideas?

Thanks!!!

Creamola, 2018-01-21:
OK, so I am making progress, after realising that no settings that I changed via the web interface were persistent, but "SetOption13 1" has now made that so.

I now have this reliably running on 5.8.0 as T1 3CH, but it still will not accept an OTA update (even to minimal firmware) and just powered up via the FDI serial interface, switches T1 to T3 randomly toggle all on their own accord.

Creamola, 2018-01-21:
Apologies for my wittering .... Finally sussed it!

Now running with 5.8.0 (I can't OTA any later versions) and all in situ as virtual switches in Domoticz. 
Turns out, that if 240V is not connected - i.e. just powered by the serial interface, then the switches seem to randomly float on/off.

geekypenguin, 2018-01-22:
There has been a warning about this on the wiki an dother support threads for a while, but I have made it more obvious on the wiki page.
Please close the support thread if the issue is resolved

closed 2018-06-05 by stale[bot]