arendst/Tasmota #1666: DS18b20 resolution

matsekberg, 2018-01-21:
It seems like the resolution is 0.5 degress C. Is there a way to get better resolution?

arendst, 2018-01-21:
????? If you mean the output you can change using TempRes command. I have no control over the physical resolution of the DS18B20.

I suggest you get familiar with available commands. See wiki.

matsekberg, 2018-01-21:
Oh sorry, i looked in the docs but not good enough. Thanx.

matsekberg, 2018-01-21:
Hmm the TempRes seems to set the number of decimal places only. The DS18b20 can be programmed from 9 up to 12 bits of resolution, can thet be configured?

arendst, 2018-01-24:
Try the ds18x20 driver which contains code to set the resolution to 12-bit. Enable with define USE_DS18x20 in user_config.h.

Once used you might want to go back to the current default driver as the DS18B20 should keep it's configured resolution setting.

closed 2018-06-08 by stale[bot]