arendst/Tasmota #1667: Button Names for Web interface

stoehrmark, 2018-01-21:
Can the "Friendly Name" be used in the web interface instead of "Toggle".

I have and Electrodragon and would like to see the 2 switches as my "Friendly Names" or just under the button put the friendly names. 

eg

⌈‾‾‾‾‾‾‾‾‾‾‾‾‾‾‾⌉    ⌈‾‾‾‾‾‾‾‾‾‾‾‾‾‾‾⌉
|&nbsp;&nbsp;&nbsp;&nbsp;TOGGLE 1&nbsp;&nbsp;&nbsp;&nbsp; |    | &nbsp;&nbsp;TOGGLE 2&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;|
⌊_______________⌋    ⌊_______________⌋

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;TO

⌈‾‾‾‾‾‾‾‾‾‾‾‾‾‾‾⌉    ⌈‾‾‾‾‾‾‾‾‾‾‾‾‾‾‾⌉
| Kids Bathroom|    |&nbsp;&nbsp;&nbsp;Kids Shower &nbsp; |
⌊_______________⌋    ⌊_______________⌋
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;OR
⌈‾‾‾‾‾‾‾‾‾‾‾‾‾‾‾⌉    ⌈‾‾‾‾‾‾‾‾‾‾‾‾‾‾‾⌉
|&nbsp;&nbsp;&nbsp;&nbsp;TOGGLE 1&nbsp;&nbsp;&nbsp;&nbsp; |    | &nbsp;&nbsp;TOGGLE 2&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;|
⌊_______________⌋    ⌊_______________⌋
&nbsp;&nbsp;  Kids Bathroom&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Kids Shower

As always thank you for all the hard work on this project.

davidelang, 2018-01-21:
this is unlikely to happen due to the added ram/code space needed. Tasmota is 
primarily a mqtt based firmware, the web interface is for setup and 
debugging. So things that make it take more resources face an uphill battle.

closed 2018-06-04 by stale[bot]

Chelovekmolekula, 2020-01-10:
Implement Please

meingraham, 2020-01-10:
Try `WebButton<x> Desc` where `<x>` is the number of the button corresponding to the relay number. `Desc` is any text... but don't carried with long button descriptions.

Chelovekmolekula, 2020-01-11:
> Попробуйте, `WebButton<x> Desc`где `<x>`находится номер кнопки, соответствующий номеру реле. `Desc`это любой текст ... но не переносится с длинными описаниями кнопок.

It worked, thank you so much!
Tasmota 8.1.0.3

Chelovekmolekula, 2020-01-11:
> Try `WebButton<x> Desc` where `<x>` is the number of the button corresponding to the relay number. `Desc` is any text... but don't carried with long button descriptions.

It worked, thank you so much!
Tasmota 8.1.0.3

lksmj, 2021-05-10:
이 게시물 덕분에 스위치 토글 이름을 설정할수 있어 큰 도움이 됐습니다.
감사합니다.~!

ascillato, 2021-05-10:
> 이 게시물 덕분에 스위치 토글 이름을 설정할수 있어 큰 도움이 됐습니다.
> 감사합니다.~!

Using Google translate:

> Thanks to this post, it was very helpful to be able to set the switch toggle name.
> Thank you~!

stiw47, 2022-01-29:
> Try `WebButton<x> Desc` where `<x>` is the number of the button corresponding to the relay number. `Desc` is any text... but don't carried away with long button descriptions.

Thank youuuuuuuuu!!!!

telece2, 2023-06-05:
Is there any way to enter multiline messages in a long button description text, so that the tasmota home page keeps a reasonable width?
I tried without success the following: %n, \n, ~0D

Thanks

sfromis, 2023-06-05:
Try with html, like `<br>`

telece2, 2023-06-09:
Yes, that worked! Thanks a lot @sfromis!