arendst/Tasmota #1668: Google Home Speaker + tasmota

marcinbauer85, 2018-01-21:
Is there a way to controll Sonoff Basic with Google Home Speaker? Since its flashed I can't add eWELink app to GH to controll the devices. So from what I've read there are two options of emulating:
1. Wemo
Problem: I need to first add 'WeMo' devices to the app/wemo account. As most IoT devices are configured at the start (so is the Sonoff with eWELink), you have to put it into AP mode, and connect your smartfone to this AP. This is a problem since when a Tasmota Sonoff is configured it connects to your wifi. Ok so put a Tasmota Sonoff into AP mode? I can, but then it gets out of WeMo emulation...

2. Hue
The Hue app finds it, but when I'm setting up the sonoff (give it a name, room) and then try to save the app gives an Error: "Save failed. Please try again"

Why flash Tasmota when I want to use Google Home Speaker? The speaker works great with voice commands (something Hass and other platforms doesn't have as good). I want to also use HomeAssistant to control my home remotely. Google Home doesn't work if you're not on your wifi, and I want to use some custom solutions from

brianbrown1962, 2018-01-22:
I set up my HASS to expose the API publicly via https. I use ifttt.com to set up triggers from my google home speakers to send messages to HASS, which sends commands to my sonoff's via MQTT. Works great.

marcinbauer85, 2018-01-23:
That is one other way out, but I'd rather not use another service in the whole chain, because of lags and dependencies on ifttt to run smootly.

jk3lly, 2018-02-21:
I use the Google assistant component for HA.

https://home-assistant.io/components/google_assistant/

Bit of a pain to set up but it works perfectly with no noticeable delay. It's also more reliable than using ewelink cloud service.