arendst/Tasmota #1669: Flash S20 v1.3 ?

ElecTriX666, 2018-01-21:
I have flashed over 8 basics without any problems but this S20 EU is not responding no matter what I do.
Tried flipping the TX / RX around but no difference.

- Soldered a 4-pin header (has worked great on the basic).
- Connected: GND TX RX VCC (from top to bottom)
- Press button for 2-3 seconds while attaching VCC
- Settings in Arduino (port is inactive because not inserted while taking screenshot)

[image]

No matter what I do I always get:

> warning: espcomm_sync failed
> error: espcomm_open failed
> error: espcomm_upload_mem failed
> error: espcomm_upload_mem failed

Anyone?

gajotnt, 2018-01-21:
Flash size 1024 (no spiffs)

ElecTriX666, 2018-01-21:
@gajotnt I dont have that option??

[image]

ElecTriX666, 2018-01-21:
Still doesn't work and same error..

gajotnt, 2018-01-21:
Are you holding GPIO 0 to ground when connecting the programer?

ElecTriX666, 2018-01-21:
@gajotnt 

Pushing down the button while connecting VCC, tried holding for 2-3 seconds then letting go but also tried holding down button constantly.

Have they changed so button is not GPIO0 any more?

gajotnt, 2018-01-21:
Can you post some pics of the board? Maybe someone can tell if there are changes.

ElecTriX666, 2018-01-21:
[image]
[image]
[image]
[image]
[image]
[image]

ElecTriX666, 2018-01-22:
"Unknown" sounds really bad..

[image]

ElecTriX666, 2018-01-22:
Right.. After a reboot it works and I have v1.5.5. OTA here I come!

xelarep, 2018-01-22:
Had a similar problem in PlatformIO - used firmware.bin and esptool:
`cd Sonoff-Tasmota-master/.pioenvs/sonoff-DE
cp ~/Downloads/esptool-2.2/esptool.py .
python esptool.py --port /dev/cu.usbserial-AH01CDYO erase_flash
python esptool.py --port /dev/cu.usbserial-AH01CDYO write_flash -fs 1MB -fm dout 0x0 firmware.bin`

gjlawran, 2018-02-17:
@xelarep Thanks worked for me - after I changed the flash size to _detect_
`esptool.py --port COM11 write_flash -fs detect -fm dout 0x0 firmware.bin`

algozzy, 2018-03-13:
Similar problem but works like a charm with the flash size detect

NicoBijl, 2018-06-02:
I have two S20 EU v1.3 here, but got issues flashing them. 
Tried to flash one, no luck, keeps restarting after the flash. 
Then connected the second one (unchanged) to the ewelink app, and it is firmware 1.6. 

Is that a issue? or can i still do wired flash? using platformio..?
Looks like the flash on the first device went all right, but maybe someone can help me with the restart issue? 

I even tried setting  WIFI_CONFIG_TOOL to WIFI_MANAGER, but can't find the wifi AP.. 

This is not my first sonoff/tasmota, but it might be the first S20 v1.3.

wb441, 2018-06-24:
I was able to flash Tasmota to my Type E S20 back to back with a 4CH Pro Rev2.
For the latter, it showed up in my network, I could enter MQTT properties & control it from HASSIO.
However, the S20 doesn't show up. Do any of you guys see my mistake? Any differences to be expected betw 4CHProR2 and S20?

[image]

domhaas, 2018-07-10:
Flashed successfully a current tasmota to a S20 v1.3 EU. Used a Pi3 for the connection. Didn't need to set the flash-size:
`esptool.py --port /dev/ttyS0 write_flash -fm dout 0x0 sonoff.bin`

Ugagen, 2018-08-14:
I had the same problem. 
Connection the wires rx-rx tx-tx solved the problem.

ascillato, 2018-08-23:
Hi,

Closing as lasts comments said that the issue is solved. Ask to reopen if needed. Thanks

closed 2018-08-23 by ascillato2