arendst/Tasmota #1672: Keeping track of all of the devices, how do you do it?

gajotnt, 2018-01-22:
Right now i have 7 devices using Tasmota, soon will be alot more, for now i can remenber all the IP for easy access to all of then, but when i get more, it will be alot harder, what do you guys use to keep track of the devices?
A document with links/names (html, word)? Or something diferent?

davidelang, 2018-01-22:
I set fixed DHCP reservations in my router and define the name in DNS to go 
along with it.

I am running LEDE on my router, which makes doing this sort of thing trivial

gajotnt, 2018-01-22:
Yes I also have the dnsnames and DHCP reservations, but still, soon it will be alot of devices to remember the names xD
Was just wondering what "y'all" used. 
Think I'm going to make a HTML file with links to the devices.

calcipher49, 2018-01-22:
I use dnsmasq as DHCP server and I fix IP address knowing mac address (using dhcp-host command) for each sonoff device.

thomasnordquist, 2018-01-22:
Use a MQTT broker which every device connects to, give every device a name and forget about their IP addresses.

I for one use mosquitto, it is very easy to set up.

MQTT allows you to send and receive status updates from your devices.

With DNS-only solutions you loose the abstraction and bi-directional communication and MQTT offers.

Worth mentioning: I believe the tasmota devices advertise their names using MDNS, so they should always have the same name, even without a dnsmask setup.

arendst, 2018-01-22:
Prefer MQTT but I do use dnsmasq too ;-)

davidelang, 2018-01-22:
On Mon, 22 Jan 2018, Thomas Nordquist wrote:

> With DNS-only solutions you loose the abstraction and bi-directional communication and MQTT offers.

why does having predictable, unique DNS names loose the bi-directional 
communications?

thomasnordquist, 2018-01-22:
Maybe I explained it badly.
With MQTT you not only have solved the naming and you get real time updates
of your device.

If for instance someone turns on your 220V socket with the hardware button,
you'll be notified instantaneously.
So you don't need to poll for changes.

(You can "subscribe" to topics, if the topic "sonoff/some_device/POWER"
changes the MQTT server sends a notification to all clients which have
subscribed to this specific topic)

gajotnt, 2018-01-22:
Simple html file with an iframe, and does the job of giving me easy acess to all my devices for when i need to update them or change any settings :)

[image]

reloxx13, 2018-01-24:
you could take a look at my [sonweb](https://github.com/reloxx13/SonWEB)

gajotnt, 2018-01-24:
Awesome, i knew There had to be something like this :)

closed 2018-06-04 by ascillato